adafruit/Adafruit_Blinka #341: Board not detected

indikaediri, 2020-09-05:
Hi

is there's a way for getting Blinka working on Banana Pi Zero M2?

Chip id:  SUN8I
Board id:  None

Is this a DragonBoard 410c? False
Is this a Pi 3B+? False
Is this a Pi 4B? False
Is this a 40-pin Raspberry Pi? False
Is this a Raspberry Pi Compute Module? False
Is this a BeagleBone Black? False
Is this a BeagleBone Green? False
Is this a Giant Board? False
Is this a Coral Edge TPU? False
Is this a SiFive Unleashed?  False
Is this a PYNQ Board? False
Is this a Rock Pi board? False
Is this a Clockwork Pi board? False
Is this an embedded Linux system? False
Is this a generic Linux PC? False
Is this a UDOO Bolt? False
Is this an ASUS Tinker Board? False
Is this an OS environment variable special case? False

-----------------------------------------------------
BOARD=bananapim2zero
BOARD_NAME="Banana Pi M2 Zero"
BOARDFAMILY=sun8i
----------------------------------------------------

model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 22.85
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : Allwinner sun8i Family
Revision        : 0000
Serial          : 02c00142d93cf6c2

thank you

makermelissa, 2020-09-07:
You'll need to add support in Adafruit_PlatformDetect. Here's a guide on how to do it: https://learn.adafruit.com/adding-a-single-board-computer-to-platformdetect-for-blinka. It looks like the processor is being detected, so that should make it easier.

indikaediri, 2020-09-08:
Thanks so much M. Managed to get the board detected

closed 2020-09-08 by indikaediri