adafruit/Adafruit_Blinka #346: How to install on tinypico?

deftio, 2020-09-30:
I have a tinypico
https://www.adafruit.com/product/4335

and am trying to connect this:
https://www.adafruit.com/product/4485

how do I bring up Adafruit_Blinka or equivalent since tinypico is micropython and won't accept circuit python?

FoamyGuy, 2020-10-01:
I don't think that you can use Adafruit_Blinka for that. The devices that support Blinka are listed here: https://circuitpython.org/blinka

Adafruit_Blinka is made for Single Board Computer devices like Raspberry Pi and others. 

Tinypico is ESP32 Microcontroller though, not a linux SBC.

deftio, 2020-10-01:
Ok got it, thanks  Any thoughts on how to quickly port their circuit python library for the Adafruit LSM6DS33 + LIS3MDL - 9 DoF IMU with Accel / Gyro / Mag - STEMMA QT Qwiic?

I started to plumb though the circuit python driver but it's littered with dependencies from adafruit busio, registers, etc.

caternuson, 2020-10-01:
Yep, Blinka is for SBC's like linked above.

More info on why no CP for ESP32 (same general issue as ESP8266):
https://learn.adafruit.com/welcome-to-circuitpython/circuitpython-for-esp8266

You can just think of `busio` and `register` as lower level libraries. To port the library, focus on the code in the main library(s) repo to see what is being done in terms of I2C comms (use of `register` may obfuscate that a bit though). And then port that to MP's equivalent of I2C comms.

closed 2020-10-01 by caternuson

tannewt, 2020-10-01:
@FoamyGuy @caternuson note that Blinka was actually started as a compatibility layer for use with MicroPython. It just had SBC support added and actually used. So, I'm not sure of the state of MicroPython support.

@deftio You can likely create an I2C bus object with MicroPython under-the-hood that you can then pass into register and bus device.

makermelissa, 2020-10-01:
I've tried to keep MP support as much as possible, but have neither tested nor heard any reports of it not working.