adafruit/Adafruit_Blinka #347: AttributeError: module 'microcontroller' has no attribute 'delay_us

olixr, 2020-10-05:
Running this circuitpython library on a Raspberry Pi 4 I am running into issues with the `microcontroller` module missing the `delay_us` function that is documented [here](https://circuitpython.readthedocs.io/en/5.3.x/shared-bindings/microcontroller/__init__.html?highlight=delay_us#microcontroller.delay_us).

Installed the library through blinka on the pi using the command `pip3 install adafruit-blinka` 

When I did digging into the `microcontroller` module using `help(microcontroller)` I get the following:

```
Help on package microcontroller:

NAME
    microcontroller - Microcontroller pins

PACKAGE CONTENTS
    pin

CLASSES
    adafruit_blinka.Enum(builtins.object)
        Pin

    class Pin(adafruit_blinka.Enum)
     |  Pin(pin_id)
     |
     |  Reference Pin object
     |
     |  Method resolution order:
     |      Pin
     |      adafruit_blinka.Enum
     |      builtins.object
     |
     |  Methods defined here:
     |
     |  __init__(self, pin_id)
     |      Identifier for pin, referencing platform-specific pin id
     |
     |  __repr__(self)
     |      Assumes instance will be found as attribute of own class.
     |      Returns dot-subscripted path to instance
     |      (assuming absolute import of containing package)
     |
     |  ----------------------------------------------------------------------
     |  Class methods inherited from adafruit_blinka.Enum:
     |
     |  iteritems() from builtins.type
     |      Inspects attributes of the class for instances of the class
     |      and returns as key,value pairs mirroring dict#iteritems
     |
     |  ----------------------------------------------------------------------
     |  Data descriptors inherited from adafruit_blinka.Enum:
     |
     |  __dict__
     |      dictionary for instance variables (if defined)
     |
     |  __weakref__
     |      list of weak references to the object (if defined)

DATA
    board_id = 'RASPBERRY_PI_4B'
    chip_id = 'BCM2XXX'
``` 

Is there something further I need to install to get all the functions listed on the documentation? Perhaps I am missing something here? I have performed updates on the library just to be sure I have all the latest versions.

olixr, 2020-10-05:
I tried to run somethings from the LCD library. I was encountering issues and when I dug into the logs it says the `delay_us` does not exist on the `microcontroller`. I read through the guides and the docs. [The docs lists the delay_us here. ](https://circuitpython.readthedocs.io/en/5.3.x/shared-bindings/microcontroller/__init__.html?highlight=delay_us#microcontroller.delay_us) However upon trying to import and access this function. Its doesn't exist. 

I have broken it down into an even simpler example that explains: 

Run this example
```
import micocontroller
microcontroller.delay_us(1)
```

Get the following result of : `AttributeError: module 'microcontroller' has no attribute 'delay_us'`

makermelissa, 2020-10-05:
Blinka is written in Python, so adding this function likely would not actually have the precision you would expect with CircuitPython, which is written in c. That being said, you could try using time.monotonic_ns() to get a before and after and that would give you something that works now.

ladyada, 2020-10-12:
closing cause it seems resolved

closed 2020-10-12 by ladyada

olixr, 2020-10-12:
The issue is still that using adafruit-circuitpython through blynka does not contain the full API as the other library. Running some various samples across the site you run into issues based on if you are running this on a Pi or circuitpython device. 

The main problem I encountered that raised this is the missing `microcontroller.delay_us(1)`. There was a call to this method in a library that worked on most devices. When trying to install on the RPi since blynka did not have this method, it caused errors. 

Perhaps a simply alias of delay_us to time.sleep would allow more compatibility when installing on the pi?

ladyada, 2020-10-12:
hi - where do you see delay_us used, what is the 'other library' and 'various samples' found?

olixr, 2020-10-12:
I was testing out a library from @dhalbert for the character lcd. He had used[ delay_us here](https://github.com/dhalbert/CircuitPython_LCD/blob/e79cdd4d0f2fa35642f4130ab869503be3122830/lcd/i2c_pcf8574_interface.py#L84). This was found when discussing the adafruit charLCD libraries. 

I was following install directions for the RPi with circuit python and this is what lead me to this. I understand that the delay_us would not work the same on the RPi I was mainly suggesting some sort of compatibility that just uses sleep under the hood. 

If this is not in the scope of this library that is fine let me know.

dhalbert, 2020-10-12:
You could just convert the `delay_us()` calls to `time.sleep()` calls. I wrote that library years ago when Blinka didn't exist, for a specific project (it's not an official Adafruit library).

olixr, 2020-10-12:
I did the conversions to use sleep already. I thought this was something under the umbrella of Adafruit. I didn't realize this was just a personal side project from you, my mistake. I am just wrapping my head around the circuit python ecosystem and its not initially designed for the RPi I understand. I will look into blynka more and other solutions as needed to fill the gap. Thanks again for the help.

makermelissa, 2021-03-05:
FYI, in case you're still interested, @dhalbert just added this function.