adafruit/Adafruit_Blinka #350: Support for Nanopi Duo2

modi12jin, 2020-10-07:
[image]

[image]

> It can be plugged into the breadboard, hoping to add its support.

modi12jin, 2020-12-31:
[image]

 > I have completed the first part

modi12jin, 2021-01-02:
[image]

> circuitPython for nanopi duo2

ladyada, 2021-01-02:
awesome, that looks great! please submit PR for platformdetect and blinka support :)

ladyada, 2021-01-03:
@modi12jin will you be sending a PR?

modi12jin, 2021-01-04:
@ladyada I posted a request, why did it fail?
I'm a novice and don't know where is the problem

closed 2021-02-09 by modi12jin