adafruit/Adafruit_Blinka #351: JETSON_XAVIER busio.py constructor attempts to import from machine

nostrumuva, 2020-10-11:
Python 3.6.9 (default, Jul 17 2020, 12:50:27) 
[GCC 8.4.0] on linux
>>> import sys
>>> from adafruit_blinka.agnostic import board_id
>>> print(board_id)
JETSON_XAVIER
>>> from adafruit_blinka.board.nvidia.jetson_xavier import *
>>> import busio
>>> print(busio.I2C(SCL,SDA))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/busio.py", line 31, in __init__
    self.init(scl, sda, frequency)
  File "/usr/local/lib/python3.6/dist-packages/busio.py", line 59, in init
    from machine import I2C as _I2C
ModuleNotFoundError: No module named 'machine'

nostrumuva, 2020-10-11:
It looks like the detector just ain't working.  I'm trying to manually create an i2c object instead
>>> for portId,portScl,portSda in i2cPorts:
...     print(portId,portScl,portSda)
... 
8 DP_AUX_CH3_P DP_AUX_CH3_N
1 GEN2_I2C_SCL GEN2_I2C_SDA

nostrumuva, 2020-10-12:
Ok, so I got the Xavier working again with a workaround... here it is:
import threading
from adafruit_servokit import ServoKit
import board
import busio
from adafruit_blinka.board.nvidia.jetson_xavier import *
from adafruit_pca9685 import PCA9685
from adafruit_blinka.microcontroller.generic_linux.i2c import I2C as _I2C
from microcontroller.pin import i2cPorts
class I2COverride(busio.I2C):
    """
    @brief override class for I2C
    """
    def __init__(self):
        self._i2c=None
        self._lock=None
...
temp_i2c=None
for temp_port_id,temp_port_scl,temp_port_sda in i2cPorts:
    try:
        if SCL==temp_port_scl and SDA==temp_port_sda:
            temp_i2c=I2COverride()
            temp_i2c._i2c=_I2C(temp_port_id,mode=_I2C.MASTER,baudrate=400000)
            temp_i2c._lock=threading.RLock()
            break
    except RuntimeError:
        pass
if temp_i2c is None:
    print("no I2C available")
else:
    self.servo_kit=ServoKit(channels=16,i2c=temp_i2c)

until the detector class is fixed for xavier, you can use this.  never actually looked into the detector class...

ladyada, 2020-10-12:
@swarren is looking at this

swarren, 2020-10-12:
Yes, https://github.com/adafruit/Adafruit_Python_PlatformDetect/pull/106 will fix the machine import issue.

In the test code in this bug, I believe that this:

`from adafruit_blinka.board.nvidia.jetson_xavier import *`

should be this:

`from adafruit_blinka.board.jetson_xavier import *`

... since there is no nvidia module in Adafruit Blinka?

With that fixed, I get the following on Jetson Nano at least, so this still doesn't work:

ubuntu@ubuntu:~$ ./test2.py 
JETSON_NANO
Traceback (most recent call last):
  File "./test2.py", line 8, in <module>
    print(busio.I2C(SCL,SDA))
  File "/home/ubuntu/Adafruit_Blinka/src/busio.py", line 16, in __init__
    self.init(scl, sda, frequency)
  File "/home/ubuntu/Adafruit_Blinka/src/busio.py", line 24, in init
    from microcontroller.pin import i2cPorts
  File "/home/ubuntu/Adafruit_Blinka/src/microcontroller/__init__.py", line 27, in <module>
    if chip_id == ap_chip.ESP8266:
AttributeError: module 'adafruit_platformdetect.chip' has no attribute 'ESP8266'
Exiting... 
Cleaning up pins

I assume this is unrelated to any Jetson code, but I'll see if I can quickly find the issue.

swarren, 2020-10-12:
Ah, I had a stale copy of Blinka. With the latest version pulled in, the test code from this bug works without modification or issue now. We can close this bug.

closed 2020-10-12 by ladyada