adafruit/Adafruit_Blinka #355: reinitialising NeoPixel library after calling deinit() does not work

AndrewMohawk, 2020-10-17:
Heya,

I need to re-initialise the NeoPixel library and it looks like I should be calling deinit() after I am done, but once I have done this, when I re-initialise the library the strip is no longer responsive:

```
>>> import board, neopixel
>>> x = neopixel.NeoPixel(board.D18,300,auto_write=False); x.fill((255,0,0)); x.show();
>>> #Strip showing red, everything is good
>>> x.deinit()
>>> #Strip is off, looks okay
>>> x = neopixel.NeoPixel(board.D18,300,auto_write=False); x.fill((0,255,0)); x.show();
>>> #No errors but strip is not responsive
```

I suspect there could be an issue in how it resets the pin, functionally the object works without any errors, but it no longer lights up the LEDs.

Thanks for all the work on the lib, its super useful!

makermelissa, 2020-10-19:
I wonder if this is related to #281.

jamesrusso, 2020-12-12:
I have tested the example code with the fix from #378 and confirmed that after running the test, my LEDs are all green (as expected). So, I think with my fix in #378 this issue is resolved. The only concern is just the requirement that both libraries (BlinkA and NeoPixel) need to be released together. I did add code to make NeoPixel safe if the method is not exposed.

makermelissa, 2020-12-23:
Fixed by #383.

closed 2020-12-23 by makermelissa