adafruit/Adafruit_Blinka #362: Using SPI TFTs with other SPI peripherals?

caternuson, 2020-11-02:
This seems to be a general point of confusion. This thread shows an example issue:
https://forums.adafruit.com/viewtopic.php?f=50&t=171326

If there is a way to use SPI TFT along with other SPI peripherals, then seems like something is overriding something with the current setup?

Or, if it's simply that SPI TFT's can not be used with other SPI peripherals, then add info on that as needed in guide(s)?

ladyada, 2020-11-02:
ya this is basically the same issue as
https://github.com/adafruit/Adafruit_Blinka/issues/329
we should, for raspi, see if one port is taken (CS0) and if so, try CS1
AND
if they try to create an SPI device with CS1, use CS1 isntead of CS0 and toggling CS1
if that makes sense :)

makermelissa, 2023-09-13:
Ok, as far as I know this was resolved, at least on Raspberry Pi. The fix is to reassign the chip enable lines. See https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/spi-sensors-devices#reassigning-the-spi-chip-enable-lines-3097985.

closed 2023-09-13 by makermelissa