adafruit/Adafruit_Blinka #364: Edge detection in Blinka

yeyeto2788, 2020-11-09:
Hey @makermelissa!

We're porting @olixr [Mudpi](https://mudpi.app/) application [code](https://github.com/mudpi/mudpi-core) to use Blinka on the core in order to allow users to use their Linux SBC https://github.com/mudpi/mudpi-core/issues/15 (previously RPI was the only supported one) which is almost done by @SecT0uch

One of the issues we encountered is that we had `RPi.GPIO` as a dependency that does edge detection for event addition and handling and we couldn't find that reference on Blinka. We did find it on Circuitpython [documentation](https://circuitpython.readthedocs.io/en/1.0.0/docs/esp8266/tutorial/pins.html#external-interrupts) but nothing on Blinka itself.

Another proposed solution was to use [adafruit_debouncer](https://circuitpython.readthedocs.io/projects/debouncer/en/latest/) for edge detection, do you think it could help?

Can you guys help us a bit on that, maybe I didn't search in the correct place 😢

ladyada, 2020-11-09:
hihi you can just use RPi.GPIO at the same time, for times you need special handling i've done it :)

closed 2020-11-09 by ladyada

SecT0uch, 2020-11-09:
Hi
Also for non-pi SBC ?

ladyada, 2020-11-09:
thats the trade off - other SBC may not have it - it depends on the chip.

SecT0uch, 2020-11-09:
Managed to do what I wanted with `fell` and `rose`  [adafruit_deboucer](https://circuitpython.readthedocs.io/projects/debouncer/en/latest/)'s methods. :+1:

yeyeto2788, 2020-11-10:
@ladyada thank you very much for the reply.

sgrobler, 2022-06-14:
@ladyada - How does one use RPi.GPIO at same time? I'm getting an error message:
```
GPIO.setmode(GPIO.BOARD)
ValueError: A different mode has already been set!
```

sgrobler, 2022-06-25:
@makermelissa - many thanks. I realised that I just need to use BCM mode and then I can use RPi.GPIO as well as Blinka.