adafruit/Adafruit_Blinka #367: How I make work the blinka sensor examples if I use i2c-0 in RPi4?

FPSychotic, 2020-11-17:
I have no idea of programing, my AMG8833, vl5xx and as72xx worked straight away in i2c-1, but I need use them connected to i2c-0, they are recognized and the ort works, I just need chosse the i2c-0 n the python code, 
I think is this line which is shared in all the examples, anyone can help me?

# for I2C use:
i2c = busio.I2C(board.SCL, board.SDA)
sensor = AS726x_I2C(i2c)

thanks and take care!!!!

ladyada, 2020-11-17:
how are you getting to i2c-0?

makermelissa, 2020-11-17:
If you need to specify the I2C bus, you could try using this library: https://github.com/adafruit/Adafruit_Python_Extended_Bus

FPSychotic, 2020-11-17:
> how are you getting to i2c-0?

Just followed a tutorial, If I remember right you just need add a line in the RPi config.txt.
I had the 3 sensors working in i2c-1 and the PCA6xxx servo controller in i2c-0 with not problem. I read I2c-0 is for hats, but you can use it with that line, I will try find it

FPSychotic, 2020-11-17:
> If you need to specify the I2C bus, you could try using this library: https://github.com/adafruit/Adafruit_Python_Extended_Bus

Thanks a lot!! Im not sure what to do with it, but I will give a try, this could be add my first lines of code!! XD.
I guess is import the library, asisng i2c(0) to I2C variable, and the previous line I wrote will use it, is that right? (More or less XD)

FPSychotic, 2020-11-17:
> how are you getting to i2c-0?
https://learn.sparkfun.com/tutorials/raspberry-pi-spi-and-i2c-tutorial/i2c-0-on-40-pin-pi-boards

This will work

FPSychotic, 2020-11-17:
> If you need to specify the I2C bus, you could try using this library: https://github.com/adafruit/Adafruit_Python_Extended_Bus

I made it work!!!! My first code modificaton!!! thank you so much!!!!
maybe you want add the possibility of use i2c-0 and this as it works, I think wil be useful! the thermal camera amg8833 example:

`"""This example is for Raspberry Pi (Linux) only!
   It will not work on microcontrollers running CircuitPython!"""

import sys
import math
import time
import numpy as np
import busio
import board
from scipy.interpolate import griddata
from colour import Color
from adafruit_extended_bus import ExtendedI2C as I2C
import adafruit_amg88xx

i2c = I2C(0)

# low range of the sensor (this will be blue on the screen)
MINTEMP = 26.0
# high range of the sensor (this will be red on the screen)
MAXTEMP = 32.0
COLORDEPTH = 1024
SENSOR = adafruit_amg88xx.AMG88XX(i2c)

# pylint: disable=invalid-slice-index
POINTS = [(math.floor(ix / 8), (ix % 8)) for ix in range(0, 64)]
GRID_X, GRID_Y = np.mgrid[0:7:32j, 0:7:32j]
# pylint: enable=invalid-slice-index

# sensor is an 8x8 grid so lets do a square
HEIGHT = 240
WIDTH = 240

# the list of colors we can choose from
BLUE = Color("indigo")
COLORS = list(BLUE.range_to(Color("red"), COLORDEPTH))

# create the array of colors
COLORS = [(int(c.red * 255), int(c.green * 255), int(c.blue * 255)) for c in COLORS]
CONSOLE_COLORS = [
    17,
    18,
    19,
    20,
    21,
    57,
    93,
    129,
    165,
    201,
    200,
    199,
    198,
    197,
    196,
    202,
    208,
    214,
    220,
]

def map_value(x_value, in_min, in_max, out_min, out_max):
    """Maps value of the temperature to color"""
    return (x_value - in_min) * (out_max - out_min) / (in_max - in_min) + out_min

def print_there(console_x, console_y, text, color):
    """ Outputs a colored text to console at coordinates """
    sys.stdout.write("\x1b7\x1b[48;5;%dm" % (color))
    sys.stdout.write("\x1b7\x1b[%d;%df%s\x1b8" % (console_x, console_y, text))

# let the sensor initialize
time.sleep(0.1)

COLOR_RANGE = 1
while True:

    # read the pixels
    PIXELS = []
    for row in SENSOR.pixels:
        PIXELS = PIXELS + row
    PIXELS = [map_value(p, MINTEMP, MAXTEMP, 0, COLORDEPTH - 1) for p in PIXELS]

    # perform interpolation
    BICUBIC = griddata(POINTS, PIXELS, (GRID_X, GRID_Y), method="cubic")

    MAXPIXEL = 0
    MINPIXEL = 0

    # draw everything
    Y_CONSOLE = 2
    for ix, row in enumerate(BICUBIC):
        x_console = 2
        for jx, pixel in enumerate(row):
            color_index = 0
            if COLOR_RANGE != 0:
                color_index = int(round((pixel - MINPIXEL) / COLOR_RANGE))
            if color_index < 0:
                color_index = 0
            if color_index > len(CONSOLE_COLORS) - 1:
                color_index = len(CONSOLE_COLORS) - 1
            print_there(x_console, Y_CONSOLE * 2 - 2, "  ", CONSOLE_COLORS[color_index])
            if pixel > MAXPIXEL:
                MAXPIXEL = pixel
            if pixel < MINPIXEL:
                MINPIXEL = pixel
            x_console += 1
        Y_CONSOLE += 1
    sys.stdout.flush()
    HEAT_RANGE = MAXPIXEL - MINPIXEL
    COLOR_RANGE = HEAT_RANGE / len(CONSOLE_COLORS)`

closed 2020-11-17 by FPSychotic