adafruit/Adafruit_Blinka #368: Board not supported exception - (Raspberry 4 and MLX90640)

Pedalbo, 2020-11-18:
Hello, I am almost new to RPI - yes, I know its about time :) - and I am not able to get the Adafruit example running! I have made the debug in the issue #66, following the nice suport from @ladyada, with this results: 

pi@raspberrypi:~ $ cat /proc/cpuinfo
`processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 126.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 126.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 126.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 126.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2711
Revision	: c03114
Serial		: 10000000fdb35e02
Model		: Raspberry Pi 4 Model B Rev 1.4
`

Also, the detect.py to test the plataform: 

`Chip id:  BCM2XXX
Board id:  None

Is this a DragonBoard 410c? False
Is this a Pi 3B+? False
Is this a Pi 4B? False
Is this a 40-pin Raspberry Pi? False
Is this a Raspberry Pi Compute Module? False
Is this a BeagleBone Black? False
Is this a BeagleBone Green? False
Is this a Giant Board? False
Is this a Coral Edge TPU? False
Is this a SiFive Unleashed?  False
Is this a PYNQ Board? False
Is this a Rock Pi board? False
Is this a Clockwork Pi board? False
Is this an embedded Linux system? True
Is this a generic Linux PC? False
Is this a UDOO Bolt? False
Is this an ASUS Tinker Board? False
Is this an STM32MP1 Board? False
Is this an OS environment variable special case? False
Raspberry Pi detected.`

Any help would be very apreciated, I am running of option here...

makermelissa, 2020-11-18:
Thanks, this rev code is a new one. I just added it in https://github.com/adafruit/Adafruit_Python_PlatformDetect/pull/119 so it should be fixed soon.

closed 2020-11-18 by ladyada

Pedalbo, 2020-11-19:
Hello. It solved the problem, but unfurtunately the example in adafruit (mlx90640_camtest.py) crashes the RPi4 all the time! Don't even have time to get the error. The example mlx90640_simpletest.py works good... any thoughts?

makermelissa, 2020-11-19:
It really depends on the error. You said you don't have time to get an error message, so I'm assuming it's rebooting or something. If not, could you explain what you mean by "crashes".

makermelissa, 2020-11-19:
Also, please open a new issue since it's unrelated to the original issue.

Pedalbo, 2020-11-26:
Ok Melissa, its at issue #371. Thank you for your help!