adafruit/Adafruit_Blinka #369: Update README.md to emphasize CPython support

dhalbert, 2020-11-18:
The very first sentence in README.md is:
>This repository contains a selection of packages mirroring the CircuitPython API on hosts running micropython.

MicroPython and CPython are mentioned together more later, but the README could probably use an updating to emphasize the current goals, which are more about CPython.

(Is the MicroPython support still viable?)

makermelissa, 2020-11-20:
As far as I've heard MicroPython support is still viable. I know there was an issue at one point and somebody let us know and it was fixed. I haven't heard of it not working since.

ladyada, 2020-11-21:
i want to maintain micropython support - micropython community needs drivers, and especially with the new microbit this is a good solution.

jerryneedell, 2021-04-17:
I have had very little success with Micropython. I may be doing it wrong. How do you use it with Micropython? What BSPs work?

ladyada, 2021-04-17:
yeah it does not work in micropython right now

dhalbert, 2021-04-17:
I'll make a separate issue for that now.

PR #444 (2021-04-17): Make clear in README Blinka works on CPython and MicroPython
Updates the README to make clear this emulation works on both CPython and MicroPython.

Fixes #369.
closed 2021-04-17 by dhalbert