adafruit/Adafruit_Blinka #370: FT232H gets wrong error thrown

danieldmulkey, 2020-11-20:
Hi there,

I'm using a fresh install of the libraries for the FT232H breakout board with Anaconda 3.7. I'm trying to use the LIS3DH with the FT232H over either SPI or I2C. After all the imports, running:
```
i2c = busio.I2C(board.SCL, board.SDA)
int1 = digitalio.DigitalInOut(board.D4)
lis3dh = adafruit_lis3dh.LIS3DH_I2C(i2c, int1=int1)
```
yields:
```
  File "C:\Users\Daniel\.spyder-py3\temp.py", line 19, in <module>
    lis3dh = adafruit_lis3dh.LIS3DH_I2C(i2c, int1=int1)

  File "C:\Users\Daniel\anaconda3\lib\site-packages\adafruit_lis3dh.py", line 332, in __init__
    super().__init__(int1=int1, int2=int2)

  File "C:\Users\Daniel\anaconda3\lib\site-packages\adafruit_lis3dh.py", line 112, in __init__
    self._int1.pull = digitalio.Pull.UP

  File "C:\Users\Daniel\anaconda3\lib\site-packages\digitalio.py", line 170, in pull
    self._pin.init(mode=Pin.IN, pull=Pin.PULL_UP)

AttributeError: type object 'Pin' has no attribute 'PULL_UP'
```

Digging around, it looks like it should be raising a ValueError from line 39 of the [corresponding pin module](https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/microcontroller/ft232h/pin.py), but instead it provides the above potentially-more-confusing error.

Please let me know if I'm doing something wrong, but I tried to be deliberate in following the instructions and looking for an explanation.

caternuson, 2020-11-20:
This is sort of a mixed issue. The FT232H does not have internal pull ups, which is why the error is being thrown. But also, the LIS3DH driver makes the assumption that the interrupt pin will have an internal pull up. There's no allowance for using a pin without internal pulls and adding your own externally.

What is your use case for using the LIS3DH interrupt?

danieldmulkey, 2020-11-20:
I was trying to adapt [this](https://learn.adafruit.com/jupyter-on-any-computer-with-circuitpython-libraries-and-mcp2221) article to instead use the FT232H and LIS3DH. I also looked at the [FT232H](https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/spi) and [LIS3DH](https://learn.adafruit.com/adafruit-lis3dh-triple-axis-accelerometer-breakout/python-circuitpython) articles as well for reference. I don't particularly need the interrupt, so if I can remove it and let it float, then that's sufficient for me.

caternuson, 2020-11-20:
For basic acceleration readings, you should be fine without the interrupt pin. Try changing your code to this:
```
i2c = busio.I2C(board.SCL, board.SDA)
lis3dh = adafruit_lis3dh.LIS3DH_I2C(i2c)
```
and see if that works for your setup.

danieldmulkey, 2020-11-21:
Ah, that did the trick. I didn't know the story about the interrupt pin.
Thank you!!

caternuson, 2020-11-21:
Cool. Glad that worked.

closed 2020-11-21 by caternuson