adafruit/Adafruit_Blinka #375: using the library with raspberry pi

TheLamia, 2020-12-06:
I am facing issues running the library on raspberry pi, as I keep getting an error for the correct pins (12,19,13) or even non PWM channels  (5,6,7) for R,G,B leds.

> TypeError: Must provide a pin, PWMOut, or PWMChannel.

In the code, this error is raised in the case which pin does not have the frequency attribute, or its type is not <class 'Pin'>. 
when I tried modifying the code I find the following type for the pins 

> <class 'adafruit_blinka.microcontroller.bcm283x.pin.Pin'>

makermelissa, 2020-12-07:
Please verify you are using the Broadcom Pin numbers (GPIOs) and not the physical pin numbers. Here's a useful diagram: https://pinout.xyz/

makermelissa, 2020-12-23:
Closing due to lack of a response. If this is still an issue, feel free to re-open.

closed 2020-12-23 by makermelissa