adafruit/Adafruit_Blinka #378: Expose neopixel_cleanup method.

jamesrusso, 2020-12-12:
We should expose the neopixel_cleanup method to the main neopixel module. As far as I can see it's only the BCM module, so should be an easy fix.

I'll submit PR shortly.

ladyada, 2020-12-12:
thanks, yes submit a PR :) that would be great

jamesrusso, 2020-12-12:
PR #379 has been sent your way. It's just a simple change to expose the cleanup method in the "main" module. However, IU could only find one "concrete" implementation of the neopixel module.

ladyada, 2020-12-12:
yay! can you see if this example now works / issue resolved?
https://github.com/adafruit/Adafruit_Blinka/issues/355

jamesrusso, 2020-12-17:
I've submitted a new PR which resolves this locally within the BlinkA module. It basically looks to see if the length of the buffer has changed, and if yes, we go ahead and re-initialize the module.

makermelissa, 2020-12-18:
Hi @jamesrusso, first of all I think this is a really good approach. I tested this and it works if the buffer length changes as you stated. However, the original problem persists if the buffer length stays the same.

This gave me an idea. As a test, I took a similar approach and saved a reference to the buffer and used as "is not" comparison which seems to fix the issue even if the buffer length stays the same. Anyway, here's the code segment I got to work if you'd like to give it a try and maybe apply it to your PR:
```
# a 'static' object that we will use to manage our PWM DMA channel
# we only support one LED strip per raspi
_led_strip = None
_buf = None

def neopixel_write(gpio, buf):
    """NeoPixel Writing Function"""
    global _led_strip  # we'll have one strip we init if its not at first
    global _buf # we save a reference to the buf, and if it changes we will cleanup and re-initalize.

    if _led_strip is None or buf is not _buf:
        # This is safe to call since it doesn't do anything if _led_strip is None
        neopixel_cleanup()

        # Create a ws2811_t structure from the LED configuration.
        # Note that this structure will be created on the heap so you
        # need to be careful that you delete its memory by calling
        # delete_ws2811_t when it's not needed.
        _led_strip = ws.new_ws2811_t()
        _buf = buf
```

makermelissa, 2020-12-23:
#383 is now merged, so closing.

closed 2020-12-23 by makermelissa