adafruit/Adafruit_Blinka #382: pyftdi.ftdi.FtdiError: UsbError: [Errno 13] Access denied (insufficient permissions)

hgetter, 2020-12-15:
I am trying to set up my adafruit FT232H device by following the instructions on [this](https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/mac-osx) page. I am receiving this error. I have installed both the D2XX driver and the VCP driver - same error. I have tried [this](https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/troubleshooting) fix without success as well. Unsure what to try next, please help! :)

Terminal print out below:

(new_env) test@tests-MacBook-Pro-2 Python_work % brew install libusb
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 2 taps (homebrew/core and homebrew/cask).
==> New Formulae
heppdt2                                  xcinfo
==> Updated Formulae
Updated 10 formulae.
==> New Casks
curiosity
==> Updated Casks
bitbar                     macx-video-converter-pro   sequel-ace
flutter                    menumeters                 snagit
graphicconverter           perimeter81                webots
itrash                     plex-media-server
lingon-x                   scrutiny

Warning: libusb 1.0.24 is already installed and up-to-date
To reinstall 1.0.24, run `brew reinstall libusb`
(new_env) test@tests-MacBook-Pro-2 Python_work % pip3 install pyftdi
Collecting pyftdi
  Using cached pyftdi-0.52.0-py3-none-any.whl (150 kB)
Collecting pyserial>=3.0
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Processing /Users/test/Library/Caches/pip/wheels/c0/8b/dc/38abb7aa0341248f20c5fb273de5740825f8b0d226eb33d1a0/pyusb-1.1.0-py3-none-any.whl
Installing collected packages: pyserial, pyusb, pyftdi
Successfully installed pyftdi-0.52.0 pyserial-3.5 pyusb-1.1.0
WARNING: You are using pip version 20.2.3; however, version 20.3.3 is available.
You should consider upgrading via the '/Users/test/Desktop/Python_work/new_env/bin/python3 -m pip install --upgrade pip' command.
(new_env) test@tests-MacBook-Pro-2 Python_work % pip3 install adafruit-blinka
Collecting adafruit-blinka
  Downloading Adafruit-Blinka-5.9.1.tar.gz (113 kB)
     |████████████████████████████████| 113 kB 14.4 MB/s 
Collecting Adafruit-PlatformDetect>=2.18.1
  Downloading Adafruit-PlatformDetect-2.23.0.tar.gz (29 kB)
Collecting Adafruit-PureIO>=1.1.7
  Downloading Adafruit_PureIO-1.1.8.tar.gz (26 kB)
Requirement already satisfied: pyftdi>=0.40.0 in ./new_env/lib/python3.9/site-packages (from adafruit-blinka) (0.52.0)
Requirement already satisfied: pyserial>=3.0 in ./new_env/lib/python3.9/site-packages (from pyftdi>=0.40.0->adafruit-blinka) (3.5)
Requirement already satisfied: pyusb>=1.0.0 in ./new_env/lib/python3.9/site-packages (from pyftdi>=0.40.0->adafruit-blinka) (1.1.0)
Using legacy 'setup.py install' for adafruit-blinka, since package 'wheel' is not installed.
Using legacy 'setup.py install' for Adafruit-PlatformDetect, since package 'wheel' is not installed.
Using legacy 'setup.py install' for Adafruit-PureIO, since package 'wheel' is not installed.
Installing collected packages: Adafruit-PlatformDetect, Adafruit-PureIO, adafruit-blinka
    Running setup.py install for Adafruit-PlatformDetect ... done
    Running setup.py install for Adafruit-PureIO ... done
    Running setup.py install for adafruit-blinka ... done
Successfully installed Adafruit-PlatformDetect-2.23.0 Adafruit-PureIO-1.1.8 adafruit-blinka-5.9.1
WARNING: You are using pip version 20.2.3; however, version 20.3.3 is available.
You should consider upgrading via the '/Users/test/Desktop/Python_work/new_env/bin/python3 -m pip install --upgrade pip' command.
(new_env) test@tests-MacBook-Pro-2 Python_work % python3
Python 3.9.0 (default, Oct 27 2020, 14:15:17) 
[Clang 12.0.0 (clang-1200.0.32.21)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/test/Desktop/Python_work/new_env/lib/python3.9/site-packages/board.py", line 145, in <module>
    from adafruit_blinka.board.ftdi_ft232h import *
  File "/Users/test/Desktop/Python_work/new_env/lib/python3.9/site-packages/adafruit_blinka/board/ftdi_ft232h.py", line 2, in <module>
    from adafruit_blinka.microcontroller.ft232h import pin
  File "/Users/test/Desktop/Python_work/new_env/lib/python3.9/site-packages/adafruit_blinka/microcontroller/ft232h/pin.py", line 71, in <module>
    D4 = Pin(4)
  File "/Users/test/Desktop/Python_work/new_env/lib/python3.9/site-packages/adafruit_blinka/microcontroller/ft232h/pin.py", line 24, in __init__
    i2c.configure("ftdi://ftdi:ft232h/1")
  File "/Users/test/Desktop/Python_work/new_env/lib/python3.9/site-packages/pyftdi/i2c.py", line 517, in configure
    frequency = self._ftdi.open_mpsse_from_url(url, **kwargs)
  File "/Users/test/Desktop/Python_work/new_env/lib/python3.9/site-packages/pyftdi/ftdi.py", line 647, in open_mpsse_from_url
    return self.open_mpsse_from_device(device, interface,
  File "/Users/test/Desktop/Python_work/new_env/lib/python3.9/site-packages/pyftdi/ftdi.py", line 768, in open_mpsse_from_device
    frequency = self._set_frequency(frequency)
  File "/Users/test/Desktop/Python_work/new_env/lib/python3.9/site-packages/pyftdi/ftdi.py", line 2307, in _set_frequency
    self.write_data(cmd)
  File "/Users/test/Desktop/Python_work/new_env/lib/python3.9/site-packages/pyftdi/ftdi.py", line 1775, in write_data
    length = self._write(data[offset:offset+write_size])
  File "/Users/test/Desktop/Python_work/new_env/lib/python3.9/site-packages/pyftdi/ftdi.py", line 2087, in _write
    raise FtdiError('UsbError: %s' % str(ex)) from None
pyftdi.ftdi.FtdiError: UsbError: [Errno 13] Access denied (insufficient permissions)

yeyeto2788, 2020-12-17:
Have you tried [this](https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/troubleshooting#pyftdi-dot-ftdi-dot-ftdierror-usberror-errno-13-access-denied-insufficient-permissions-3041751-28) Seems like issuing `brew remove libftdi` will do the trick.

hgetter, 2020-12-17:
Hi Yeyeto,

I have tried that fix. The same Access denied error is produced.

yeyeto2788, 2020-12-17:
I'm not a Mac user but what about `sudo`? Is there a way to give your user permissions to use a device? Does the board works with other software?

makermelissa, 2020-12-23:
This is an issue in PyFTDI, so there's nothing to fix in this library.

According to https://github.com/eblot/pyftdi/issues/94, here are some things to try:
http://eblot.github.io/pyftdi/troubleshooting.html#error-access-denied-insufficient-permissions

closed 2020-12-23 by makermelissa

saepien, 2022-09-07:
Same issue... any fix yet?

makermelissa, 2022-09-07:
> Same issue... any fix yet?

Unfortunately the issue exists in code outside of our control. See https://github.com/adafruit/Adafruit_Blinka/issues/382#issuecomment-750373660. It looks like the PR I linked to was merged so maybe file another issue over there: https://github.com/eblot/pyftdi/issues/94

brainelectronics, 2023-01-31:
@hgetter @makermelissa @saepien 
I had the same issue with a FT232H used for I2C communication on my Mac
I solved it by

1. connecting the device to my Mac
2. `sudo kextunload -v -bundle com.apple.driver.AppleUSBFTDI`, according to [PyFtdi Troubleshooting](https://eblot.github.io/pyftdi/troubleshooting.html#error-access-denied-insufficient-permissions)
3. `sudo python`
```python
from pyftdi.i2c import I2cController
i2c = I2cController()
i2c.configure('ftdi://ftdi:232h/1')
# I did not get beyond this point previously
slave = i2c.get_port(0x21)
slave.exchange([0x04], 1)
```

makermelissa, 2023-01-31:
Thanks. That info should probably be added to https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/troubleshooting#pyftdi-dot-ftdi-dot-ftdierror-usberror-errno-13-access-denied-insufficient-permissions-3041751

makermelissa, 2023-01-31:
Guide updated.