adafruit/Adafruit_Blinka #387: NotImplementedError("Board not supported {}".format(board_id))

pwaleka, 2021-01-04:
Hello everyone

I am facing the same (or similar) issue like those reported here
https://github.com/adafruit/Adafruit_Blinka/issues/179
and here 
https://github.com/adafruit/Adafruit_Blinka/issues/66

Basically when running simple script I get the following Error

File "/home/pi/code_scanner/gate-enabler-pixels.py", line 3, in <module>
    import board
  File "/usr/local/lib/python3.7/dist-packages/board.py", line 175, in <module>
    raise NotImplementedError("Board not supported {}".format(board_id))
NotImplementedError: Board not supported None

Output of the detect.py script:
python3 detect.py
Chip id:  BCM2XXX
Board id:  RASPBERRY_PI_4B

Is this a DragonBoard 410c? False
Is this a Pi 3B+? False
Is this a Pi 4B? True
Is this a 40-pin Raspberry Pi? True
Is this a Raspberry Pi Compute Module? False
Is this a BeagleBone Black? False
Is this a BeagleBone Green? False
Is this a Giant Board? False
Is this a Coral Dev Board? False
Is this a Coral Dev Board Mini? False
Is this a SiFive Unleashed?  False
Is this a PYNQ Board? False
Is this a Rock Pi board? False
Is this a Clockwork Pi board? False
Is this an embedded Linux system? True
Is this a generic Linux PC? False
Is this a UDOO Bolt? False
Is this an ASUS Tinker Board? False
Is this an STM32MP1 Board? False
Is this an OS environment variable special case? False
Raspberry Pi detected.

Output of the cpuinfo is following

cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2711
Revision        : b03114
Serial          : 100000009cc88b1d
Model           : Raspberry Pi 4 Model B Rev 1.4

I tried adding b03114 to the list of revisions in Boards.py like described in https://github.com/adafruit/Adafruit_Blinka/issues/179 however I am running another issue

Traceback (most recent call last):
  File "/home/pi/code_scanner/gate-enabler-pixels.py", line 66, in <module>
    switchOffPixels()
  File "/home/pi/code_scanner/gate-enabler-pixels.py", line 37, in switchOffPixels
    pixels.show()
  File "/usr/local/lib/python3.7/dist-packages/adafruit_pypixelbuf.py", line 211, in show
    return self._transmit(self._post_brightness_buffer)
  File "/usr/local/lib/python3.7/dist-packages/neopixel.py", line 167, in _transmit
    neopixel_write(self.pin, buffer)
  File "/usr/local/lib/python3.7/dist-packages/neopixel_write.py", line 24, in neopixel_write
    return _neopixel.neopixel_write(gpio, buf)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/bcm283x/neopixel.py", line 74, in neopixel_write
    "ws2811_init failed with code {0} ({1})".format(resp, message)
RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)
swig/python detected a memory leak of type 'ws2811_t *', no destructor found. 

Thanks in advance for any hint here

caternuson, 2021-01-04:
That rev code is in the latest version of platform detect:
https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/168ff6b66548e8873d5e05ecbe4f24d2e8b22124/adafruit_platformdetect/constants/boards.py#L395
so you shouldn't have need to add it. Are you using the latest version of the library?

ladyada, 2021-01-04:
https://www.activestate.com/resources/quick-reads/how-to-update-all-python-packages/
is handy for updating all libraries!

pwaleka, 2021-01-04:
The latest version(s) of the libraries I have got installed are following:
pi@raspberrypi:~/code_scanner/tools $ pip3 list
Package                           Version
--------------------------------- -----------
Adafruit-Blinka                   5.9.2
adafruit-circuitpython-neopixel   6.0.0
adafruit-circuitpython-pypixelbuf 2.2.0
Adafruit-PlatformDetect           2.23.0
Adafruit-PureIO                   1.1.8

caternuson, 2021-01-04:
Did you undo any of the manual edits you mentioned doing? Like changing `board.py`.

What happens if you run the Blinka test?
<https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi#blinka-test-3030038-17>

pwaleka, 2021-01-04:
> Did you undo any of the manual edits you mentioned doing? Like changing `board.py`.
> 
> What happens if you run the Blinka test?
> https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi#blinka-test-3030038-17

All changes were undo'ed. However blinkatest.py is showing the following:
pi@raspberrypi:~/code_scanner $ python3 blinkatest.py
Hello blinka!
Digital IO ok!
I2C ok!
Traceback (most recent call last):
  File "blinkatest.py", line 16, in <module>
    spi = busio.SPI(board.SCLK, board.MOSI, board.MISO)
  File "/home/pi/.local/lib/python3.7/site-packages/busio.py", line 186, in __init__
    self._spi = _SPI(portId)
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_blinka/microcontroller/generic_linux/spi.py", line 22, in __init__
    self._spi = spi.SPI(device=(portid, 0))
  File "/home/pi/.local/lib/python3.7/site-packages/Adafruit_PureIO/spi.py", line 167, in __init__
    raise IOError("{} does not exist".format(device))
OSError: /dev/spidev0.0 does not exist

caternuson, 2021-01-04:
Enable SPI:
https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi#enable-i2c-and-spi-2993390-7

caternuson, 2021-01-15:
@pwaleka Is this issue resolved?

caternuson, 2021-01-19:
Closing due to lack of response.

closed 2021-01-19 by caternuson