adafruit/Adafruit_Blinka #389: Remove/insert ftdi_ft232h device

thijstriemstra, 2021-01-07:
After initializing an ftdi_ft232h device with some LEDs and then removing the USB device, causing the following error:

```
Traceback (most recent call last):
  File "hardware_ctl.py", line 70, in check_btn
    reading = button.value
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/digitalio.py", line 151, in value
    return self._pin.value() == 1
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/adafruit_blinka/microcontroller/ft232h/pin.py", line 52, in value
    current = Pin.ft232h_gpio.read(with_output=True)
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/pyftdi/i2c.py", line 331, in read
    return self._controller.read_gpio(with_output)
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/pyftdi/i2c.py", line 890, in read_gpio
    data = self._read_raw(self._wide_port)
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/pyftdi/i2c.py", line 992, in _read_raw
    self._ftdi.write_data(cmd)
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/pyftdi/ftdi.py", line 1775, in write_data
    length = self._write(data[offset:offset+write_size])
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/pyftdi/ftdi.py", line 2087, in _write
    raise FtdiError('UsbError: %s' % str(ex)) from None
pyftdi.ftdi.FtdiError: UsbError: [Errno 19] No such device (it may have been disconnected)
```

This is fine, but I'd like to re-initialize the ftdi_ft232h once it's re-inserted into the USB port.

I wrote some pyudev code that detects insert/removal of the ftdi_ft232h, so that part's sorted, but my question is: how do I re-initialize ftdi_ft232h so my python program doesn't have to be restarted?

caternuson, 2021-01-07:
Try recreating your `pin` objects used to control the LED, which should cause the FT232H configuration stuff to be re-run, since it lives in the `pin` constructor:
https://github.com/adafruit/Adafruit_Blinka/blob/084beca285f3f53e13fca734953d83ceff3b4f3c/src/adafruit_blinka/microcontroller/ft232h/pin.py#L15-L25

thijstriemstra, 2021-01-07:
Thanks for the feedback. When I re-insert the device, this code runs:

```
# inserted device
logger.debug('Inserted FT232H device')

self.device_connected = True

self.led = digitalio.DigitalInOut(board.C0)
self.button = digitalio.DigitalInOut(board.C1)
self.led.value = True
```

With the following traceback:

```
2021-01-07 20:33:02,571 - Inserted FT232H device
Exception in thread usb-monitor:
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.7/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "hardware_ctl.py", line 87, in _usb_monitor
    self.led.value = True
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/digitalio.py", line 158, in value
    raise AttributeError("Not an output")
AttributeError: Not an output
```

I'll take a look at your suggestion.

caternuson, 2021-01-07:
Be sure to set pin direction.
```python
self.led.direction = digitalio.Direction.OUTPUT
```

thijstriemstra, 2021-01-08:
Thanks for the feedback. Adding the direction helps but trying to change the LED afterwards:

```python
self.led = digitalio.DigitalInOut(board.C0)
self.led.direction = digitalio.Direction.OUTPUT
self.button = digitalio.DigitalInOut(board.C1)
self.button.direction = digitalio.Direction.INPUT
self.led.value = True
```

results in this traceback:

```
2021-01-08 16:03:03,701 - Inserted FT232H device
Exception in thread usb-monitor:
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.7/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "hardware_ctl.py", line 95, in _usb_monitor
    self.led.direction = digitalio.Direction.OUTPUT
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/digitalio.py", line 140, in direction
    self.value = False
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/digitalio.py", line 156, in value
    self._pin.value(1 if val else 0)
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/adafruit_blinka/microcontroller/ft232h/pin.py", line 53, in value
    current = Pin.ft232h_gpio.read(with_output=True)
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/pyftdi/i2c.py", line 331, in read
    return self._controller.read_gpio(with_output)
  File "/home/thijs/.virtualenvs/tmobile_captcha/lib/python3.7/site-packages/pyftdi/i2c.py", line 890, in read_gpio
    data = self._read_raw(self._wide_port)
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/pyftdi/i2c.py", line 992, in _read_raw
    self._ftdi.write_data(cmd)
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/pyftdi/ftdi.py", line 1775, in write_data
    length = self._write(data[offset:offset+write_size])
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/pyftdi/ftdi.py", line 2087, in _write
    raise FtdiError('UsbError: %s' % str(ex)) from None
pyftdi.ftdi.FtdiError: UsbError: [Errno 19] No such device (it may have been disconnected)
```

thijstriemstra, 2021-01-08:
Trying to re-configure the `C0` pin manually:

```python
i2c = I2cController()
i2c.configure("ftdi:///1")
board.C0.ft232h_gpio = i2c.get_gpio()

self.led = digitalio.DigitalInOut(board.C0)
self.led.direction = digitalio.Direction.OUTPUT
#self.button = digitalio.DigitalInOut(board.C1)
#self.button.direction = digitalio.Direction.INPUT
self.led.value = True
```

Results in a slightly different traceback:

```
Exception in thread usb-monitor:
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.7/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "hardware_ctl.py", line 87, in _usb_monitor
    i2c.configure("ftdi:///1")
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/pyftdi/i2c.py", line 517, in configure
    frequency = self._ftdi.open_mpsse_from_url(url, **kwargs)
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/pyftdi/ftdi.py", line 645, in open_mpsse_from_url
    devdesc, interface = self.get_identifiers(url)
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/pyftdi/ftdi.py", line 421, in get_identifiers
    cls.DEFAULT_VENDOR)
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/pyftdi/usbtools.py", line 333, in parse_url
    default_vendor)
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/pyftdi/usbtools.py", line 437, in enumerate_candidates
    devices = cls.find_all(vps)
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/pyftdi/usbtools.py", line 121, in find_all
    description = UsbTools.get_string(dev, dev.iProduct)
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/pyftdi/usbtools.py", line 575, in get_string
    return usb_get_string(device, stridx)
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/usb/util.py", line 322, in get_string
    langid
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/usb/control.py", line 175, in get_descriptor
    data_or_wLength = desc_size)
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/usb/core.py", line 1077, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/usb/backend/libusb1.py", line 901, in ctrl_transfer
    timeout))
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 19] No such device (it may have been disconnected)
```

But running `lsusb` in a separate terminal shows its there:

```
Bus 001 Device 008: ID 0403:6014 Future Technology Devices International, Ltd FT232H Single HS USB-UART/FIFO IC
```

thijstriemstra, 2021-01-08:
I found a related ticket (https://github.com/eblot/pyftdi/issues/187) that uses a Adafruit FT232H.

Fortunately they also wrote a few tests that look a lot like my usecase: https://github.com/eblot/pyftdi/blob/master/pyftdi/tests/ftdi.py

caternuson, 2021-01-08:
It looks like using `UsbTools.flush_cache()` is on the right track for dealing with this. What stack trace are you getting from the first example above?

thijstriemstra, 2021-01-08:
```
Exception in thread usb-monitor:
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.7/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "hardware_ctl.py", line 102, in _usb_monitor
    self.led.direction = digitalio.Direction.OUTPUT
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/digitalio.py", line 140, in direction
    self.value = False
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/digitalio.py", line 156, in value
    self._pin.value(1 if val else 0)
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/adafruit_blinka/microcontroller/ft232h/pin.py", line 53, in value
    current = Pin.ft232h_gpio.read(with_output=True)
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/pyftdi/i2c.py", line 331, in read
    return self._controller.read_gpio(with_output)
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/pyftdi/i2c.py", line 890, in read_gpio
    data = self._read_raw(self._wide_port)
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/pyftdi/i2c.py", line 992, in _read_raw
    self._ftdi.write_data(cmd)
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/pyftdi/ftdi.py", line 1775, in write_data
    length = self._write(data[offset:offset+write_size])
  File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/pyftdi/ftdi.py", line 2087, in _write
    raise FtdiError('UsbError: %s' % str(ex)) from None
pyftdi.ftdi.FtdiError: UsbError: [Errno 19] No such device (it may have been disconnected)
```

caternuson, 2021-01-08:
What if you put it in a loop similar to https://github.com/eblot/pyftdi/issues/148#issuecomment-566056995

Something like this:
```python
timeout = now() + 5.0
while now() < timeout:
    try:
        self.led = digitalio.DigitalInOut(board.C0)
        self.led.direction = digitalio.Direction.OUTPUT
        break
    except UsbToolsError:
        UsbTools.flush_cache()
        sleep(0.05)
        continue
```

thijstriemstra, 2021-01-09:
That produces the same error, only later. It seems things would be easier to debug/setup if the i2c instance would be available. Currently only the ports are available (i2c.get_port()) and there's no way to access a gpio port's parent i2c controller afaik.

caternuson, 2021-01-09:
OK, I'll try and take a look. It may require changes at the library level. Doesn't seem like there's any easy way to do this with current lib from user code.

thijstriemstra, 2021-01-09:
I managed to fix it with this user code:

```python
# inserted device
logger.debug('Inserted FT232H device')

self.device_connected = True

UsbTools.flush_cache()
time.sleep(0.01)

board.C0.i2c.configure("ftdi:///1")

self.led = digitalio.DigitalInOut(board.C0)
self.led.direction = digitalio.Direction.OUTPUT
self.button = digitalio.DigitalInOut(board.C1)
self.button.direction = digitalio.Direction.INPUT
self.led.value = True
```

And this change this to `Adafruit_Blinka/src/adafruit_blinka/microcontroller/ft232h/pin.py`:

```python
class Pin:
    """A basic Pin class for use with FT232H."""

    IN = 0
    OUT = 1
    LOW = 0
    HIGH = 1

    i2c = None
    ft232h_gpio = None

    def __init__(self, pin_id=None):
        # setup GPIO controller if not done yet
        # use one provided by I2C as default
        if not Pin.ft232h_gpio:
            # pylint: disable=import-outside-toplevel
            from pyftdi.i2c import I2cController

            # pylint: enable=import-outside-toplevel

            Pin.i2c = I2cController()
            Pin.i2c.configure("ftdi://ftdi:ft232h/1")
            Pin.ft232h_gpio = Pin.i2c.get_gpio()
```

where storing the i2c instance in the pin helps to reset the device.

thijstriemstra, 2021-01-12:
Another issue to consider is starting the program before inserting the device. It'll throw this error:

```
jan 12 10:28:07 foo python[1423]: Traceback (most recent call last):
jan 12 10:28:07 foo python[1423]:   File "src/python/hardware_ctl.py", line 12, in <module>
jan 12 10:28:12 foo python[1423]:     import board
jan 12 10:28:12 foo python[1423]:   File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/board.py", line 33, in <module>
jan 12 10:28:12 foo python[1423]:     from adafruit_blinka.agnostic import board_id, detector
jan 12 10:28:12 foo python[1423]:   File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/adafruit_blinka/agnostic/__init__.py", line 18, in <module>
jan 12 10:28:12 foo python[1423]:     chip_id = detector.chip.id
jan 12 10:28:12 foo python[1423]:   File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/adafruit_platformdetect/chip.py", line 81, in id
jan 12 10:28:12 foo python[1423]:     "BLINKA_FT232H environment variable "
jan 12 10:28:12 foo python[1423]: RuntimeError: BLINKA_FT232H environment variable set, but no FT232H device found
```

How to recover from this error without restarting the program?

caternuson, 2021-01-12:
I think that one could be handled with a `try/except` in user code.

thijstriemstra, 2021-01-13:
> I think that one could be handled with a try/except in user code.

Can you show how? Catching the error is no problem:

```python
try:
    import board
    import digitalio
except RuntimeError as e:
    if str(e) == "BLINKA_FT232H environment variable set, but no FT232H device found":
        print("No FT232H device found at startup.")
```

But this results in a missing/broken module:

```
jan 13 13:28:56 foo python[20704]: Traceback (most recent call last):
jan 13 13:28:56 foo python[20704]:   File "/usr/local/lib/python3.7/threading.py", line 926, in _bootstrap_inner
jan 13 13:28:56 foo python[20704]:     self.run()
jan 13 13:28:56 foo python[20704]:   File "/usr/local/lib/python3.7/threading.py", line 870, in run
jan 13 13:28:56 foo python[20704]:     self._target(*self._args, **self._kwargs)
jan 13 13:28:56 foo python[20704]:   File "src/python/hardware_ctl.py", line 63, in start_usb_monitor
jan 13 13:28:56 foo python[20704]:     loop.run_until_complete(self._usb_monitor(loop))
jan 13 13:28:56 foo python[20704]:   File "/usr/local/lib/python3.7/asyncio/base_events.py", line 583, in run_until_complete
jan 13 13:28:56 foo python[20704]:     return future.result()
jan 13 13:28:56 foo python[20704]:   File "src/python/hardware_ctl.py", line 92, in _usb_monitor
jan 13 13:28:56 foo python[20704]:     await self._gpio_monitor()
jan 13 13:28:56 foo python[20704]:   File "src/python/hardware_ctl.py", line 122, in _gpio_monitor
jan 13 13:28:56 foo python[20704]:     self.led = digitalio.DigitalInOut(board.C0)
jan 13 13:28:56 foo python[20704]: NameError: name 'digitalio' is not defined
```

caternuson, 2021-01-13:
Yep. Something like your second approach above. This is something for which the solution is use case specific. From the view point of this library, if there is no FT232H found, then there is nothing it can do.

thijstriemstra, 2021-01-13:
> if there is no FT232H found, then there is nothing it can do.

True but including the `flush_cache` call in the `adafruit_platformdetect/chip.py` code would help users a lot.

thijstriemstra, 2021-01-13:
Seems like this a gift that keeps on giving ;)

After waking up Ubuntu from 'suspend' the application stopped working. Traceback:

```
jan 13 18:53:37 foo python[36840]: Exception in thread usb-monitor:
jan 13 18:53:37 foo python[36840]: Traceback (most recent call last):
jan 13 18:53:37 foo python[36840]:   File "/usr/local/lib/python3.7/threading.py", line 926, in _bootstrap_inner
jan 13 18:53:37 foo python[36840]:     self.run()
jan 13 18:53:37 foo python[36840]:   File "/usr/local/lib/python3.7/threading.py", line 870, in run
jan 13 18:53:37 foo python[36840]:     self._target(*self._args, **self._kwargs)
jan 13 18:53:37 foo python[36840]:   File "src/python/hardware_ctl.py", line 79, in start_usb_monitor
jan 13 18:53:37 foo python[36840]:     loop.run_until_complete(self._usb_monitor(loop))
jan 13 18:53:37 foo python[36840]:   File "/usr/local/lib/python3.7/asyncio/base_events.py", line 583, in run_until_complete
jan 13 18:53:37 foo python[36840]:     return future.result()
jan 13 18:53:37 foo python[36840]:   File "src/python/hardware_ctl.py", line 117, in _usb_monitor
jan 13 18:53:37 foo python[36840]:     await self._gpio_monitor()
jan 13 18:53:37 foo python[36840]:   File "src/python/hardware_ctl.py", line 157, in _gpio_monitor
jan 13 18:53:37 foo python[36840]:     reading = self.button.value
jan 13 18:53:37 foo python[36840]:   File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/digitalio.py", line 151, in value
jan 13 18:53:37 foo python[36840]:     return self._pin.value() == 1
jan 13 18:53:37 foo python[36840]:   File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/adafruit_blinka/microcontroller/ft232h/pin.py", line 53, in value
jan 13 18:53:37 foo python[36840]:     current = Pin.ft232h_gpio.read(with_output=True)
jan 13 18:53:37 foo python[36840]:   File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/pyftdi/i2c.py", line 331, in read
jan 13 18:53:37 foo python[36840]:     return self._controller.read_gpio(with_output)
jan 13 18:53:37 foo python[36840]:   File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/pyftdi/i2c.py", line 890, in read_gpio
jan 13 18:53:37 foo python[36840]:     data = self._read_raw(self._wide_port)
jan 13 18:53:37 foo python[36840]:   File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/pyftdi/i2c.py", line 996, in _read_raw
jan 13 18:53:37 foo python[36840]:     raise I2cIOError('Cannot read GPIO')
jan 13 18:53:37 foo python[36840]: pyftdi.i2c.I2cIOError: Cannot read GPIO
```

thijstriemstra, 2021-02-09:
> And replaced Adafruit_Blinka/src/adafruit_blinka/microcontroller/ft232h/pin.py with:

Updated to latest blinka just now and had to apply my manual fixes again. Would you accept a pull request for these i2c changes instead @caternuson?

thijstriemstra, 2021-02-09:
> Would you accept a pull request for these i2c changes instead

See #417

caternuson, 2021-05-05:
Are you still using udev as part of this? Curious how you are actually tying udev into your Python code so it communicates that the device was plugged back in.

thijstriemstra, 2021-05-05:
Here's the full script:

```python
# Copyright (c) 2021 Thijs Triemstra

import sys
import asyncio
import logging
import threading
import subprocess
from pathlib import Path

import pyudev
import systemd.daemon
import simpleaudio as sa
from pyftdi.i2c import I2cIOError
from pyftdi.ftdi import FtdiError
from pyftdi.usbtools import UsbTools

device_found = False
try:
    import board
    import digitalio
    device_found = True
except RuntimeError as e:
    board = None
    digitalio = None

    if str(e) == "BLINKA_FT232H environment variable set, but no FT232H device found":
        print("No FT232H device found at startup.")
    else:
        raise

logger = logging.getLogger(__name__)

class Application(object):
    """
    """
    board = None
    digitalio = None

    def __init__(self, base_dir, device_connected=False, device_address='ftdi:///1'):
        """
        :type base_dir: pathlib.Path
        :type device_connected: bool
        :type device_address: str
        """
        self.base_dir = base_dir
        self.device_address = device_address
        self.lastState = False
        self.device_connected = device_connected

        # setup USB monitor
        self.usbMonitor = threading.Thread(
            name='usb-monitor',
            target=self.start_usb_monitor
        )

    def start(self):
        """
        """
        self.usbMonitor.setDaemon(True)
        self.usbMonitor.start()
        try:
            self.usbMonitor.join()
        except KeyboardInterrupt:
            sys.exit(0)

    def start_usb_monitor(self):
        """
[... 166 lines cut ...]
    # setup logging
    log_fmt = '%(asctime)s - %(message)s'
    if is_systemd:
        log_fmt = '%(message)s'
    logging.basicConfig(level=logging.DEBUG, format=log_fmt)

    # silence logger
    logging.getLogger('asyncio').setLevel(logging.WARNING)

    # start application
    base_dir = Path(__file__).parent.parent.parent.resolve()
    app = Application(base_dir, device_found)

    # store modules
    app.board = board
    app.digitalio = digitalio

    # let's go
    app.start()

if __name__ == "__main__":
    main()
```

It's not perfect but it works for me.

thijstriemstra, 2021-05-05:
I could add it as an example?

makermelissa, 2024-01-11:
Closing for now as mentioned in https://github.com/adafruit/Adafruit_Blinka/pull/417#issuecomment-1887454808

closed 2024-01-11 by makermelissa

siberianfox, 2024-12-20:
I am still having this issue, is it fixed?