adafruit/Adafruit_Blinka #391: MCP2221 I2C Clock Default Not Setting

caternuson, 2021-01-14:
Quick check just looking at clock signal during a scan:
```python
$ python3
Python 3.6.9 (default, Oct  8 2020, 12:12:24) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
>>> i2c = board.I2C()
>>> i2c.try_lock()
True
>>> i2c.scan()
[96]
```
[image]

Hmmm, more like 400k. OK, let's grab the MCP, set it again, and try again:
```python
>>> mcp = i2c._i2c._mcp2221
>>> mcp.i2c_configure(100000)
>>> i2c.scan()
[96]
```
[image]

No change.

The current code in `i2c_configure` simply blasts out the setting. It does no checking of current MCP I2C state nor if the setting actually occurred. So...what is the current state?
```python
>>> mcp._i2c_state()
37
```
It's non-zero for some reason. Let's try an i2c cancel first.
```python
>>> mcp._i2c_cancel()
>>> mcp._i2c_state()
0
>>> mcp.i2c_configure(100000)
>>> i2c.scan()
[96]
>>> 
```
[image]

There we go. Not sure if this is new or has always been there? Been too long since the initial work on this.

Discovered this behavior while sleuthing #262

ladyada, 2021-01-14:
probably always been there, we've likely been running it at 400khz forever

caternuson, 2021-01-14:
Maybe it's this?
https://github.com/adafruit/Adafruit_Blinka/blob/084beca285f3f53e13fca734953d83ceff3b4f3c/src/busio.py#L30
Should we change that to 100k?

ladyada, 2021-01-14:
yah

caternuson, 2021-01-14:
With #393 , the clock freq is now 100kHz out of the box. So that behavior is notionally fixed. But there still seems to be an issue changing it with `i2c_configure`. Issue name changed to reflect that.

ladyada, 2021-01-14:
k0ol

caternuson, 2021-01-15:
@ladyada Do you think we even need this to be available from user space? Native CP doesn't allow changing freq after instantiating. I'm thinking of just adding a leading `_` and calling it good. It works as expected through the constructor. 

Examples (junk writes just to generate SCL output):

**400kHz**
```python
(blinka) user@linux:~$ python3
Python 3.6.9 (default, Oct  8 2020, 12:12:24) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
i>>> import busio
>>> i2c = busio.I2C(board.SCL, board.SDA, frequency=400000)
>>> i2c.writeto(0x23, bytes([0x00]))
```
[image]

**50kHz**
```python
(blinka) user@linux:~$ python3
Python 3.6.9 (default, Oct  8 2020, 12:12:24) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board
>>> import busio
>>> i2c = busio.I2C(board.SCL, board.SDA, frequency=50000)
>>> i2c.writeto(0x23, bytes([0x00]))
```
[image]

ladyada, 2021-01-15:
having it set on instantiation is fine. technically, you need to have the lowest common frequency because if you have a 100khz i2c device on a bus with 1mhz i2c device, you cannot use anything higher than 100k

caternuson, 2021-01-15:
okie. im just gonna PR that simple change. i'm sort of remembering questioning this before. it's only called in the constructor and i think i was maybe even considering just refactoring it out.

we can reopen later if some use case warrants this feature.

caternuson, 2021-01-15:
Closing. See #394 .

closed 2021-01-15 by caternuson