adafruit/Adafruit_Blinka #396: "RuntimeError: I2C slave address was NACK'd" with SCD30 on MCP2221A breakout

rpavlik, 2021-01-19:
Might be related to #262 but comment there said different boards should have different issues.

I'm using the sample from the SCD30 library docs:

```python

import board
import busio
import adafruit_scd30

i2c = busio.I2C(board.SCL, board.SDA)
scd = adafruit_scd30.SCD30(i2c)

while True:
    # since the measurement interval is long (2+ seconds) we check for new data before reading
    # the values, to ensure current readings.
    if scd.data_available:
        print("Data Available!")
        print("eCO2:", scd.eCO2, "PPM")
        print("Temperature:", scd.temperature, "degrees C")
        print("Humidity:", scd.relative_humidity, "%%rH")
        print("")
        print("Waiting for new data...")
        print("")

    time.sleep(0.5)
```

Windows 10 2004, Adafruit MCP2221 breakout, Adafruit SCD30 breakout, Python 3.9.1 (installed with Scoop). The SCD30 works fine on the QT-Py I have here.

```none
ryanp@RYZENSHINE-WIN   venv  E:\src-ssd\scd30                                                                                                                                                                                                   [09:57]
❯ $env:BLINKA_MCP2221=1
ryanp@RYZENSHINE-WIN   venv  E:\src-ssd\scd30                                                                                                                                                                                                   [09:57]
❯ & e:/src-ssd/scd30/venv/Scripts/python.exe e:/src-ssd/scd30/test.py
Traceback (most recent call last):
  File "e:\src-ssd\scd30\test.py", line 8, in <module>
    scd = adafruit_scd30.SCD30(i2c)
  File "E:\src-ssd\scd30\venv\lib\site-packages\adafruit_scd30.py", line 66, in __init__
    self.measurement_interval = 2
  File "E:\src-ssd\scd30\venv\lib\site-packages\adafruit_scd30.py", line 93, in measurement_interval  
    self._send_command(_CMD_SET_MEASUREMENT_INTERVAL, value)
  File "E:\src-ssd\scd30\venv\lib\site-packages\adafruit_scd30.py", line 220, in _send_command        
    i2c.write(self._buffer, end=end_byte)
  File "E:\src-ssd\scd30\venv\lib\site-packages\adafruit_bus_device\i2c_device.py", line 102, in write
    self.i2c.writeto(self.device_address, buf, start=start, end=end)
  File "E:\src-ssd\scd30\venv\lib\site-packages\busio.py", line 115, in writeto
    return self._i2c.writeto(address, memoryview(buffer)[start:end], stop=stop)
  File "E:\src-ssd\scd30\venv\lib\site-packages\adafruit_blinka\microcontroller\mcp2221\i2c.py", line 19, in writeto
    self._mcp2221.i2c_writeto(address, buffer, start=start, end=end)
  File "E:\src-ssd\scd30\venv\lib\site-packages\adafruit_blinka\microcontroller\mcp2221\mcp2221.py", line 315, in i2c_writeto
    self._i2c_write(0x90, address, buffer, start, end)
  File "E:\src-ssd\scd30\venv\lib\site-packages\adafruit_blinka\microcontroller\mcp2221\mcp2221.py", line 233, in _i2c_write
    raise RuntimeError("I2C slave address was NACK'd")
RuntimeError: I2C slave address was NACK'd
```

ladyada, 2021-01-19:
i wonder if its because the computer is so much faster, there was an inadvertant delay between some commands. do you want to try editing adafruit_scd30.py to add some time.sleep(0.01)'s in the init code?

rpavlik, 2021-01-19:
Hmm. even `sleep(1)` between `self.reset()` and `self.measurement_interval = 2` doesn't help.

ladyada, 2021-01-19:
what if you dont reset?

rpavlik, 2021-01-19:
Then it works! 

```
Data Available!
eCO2: 922.3878173828125 PPM
Temperature: 24.009307861328125 degrees C
Humidity: 27.9052734375 %%rH

Waiting for new data...
```

All I did was comment out the self.reset()

ladyada, 2021-01-19:
hmm ok let me move this to the SCD30 library because it at least lives there
@caternuson do you have an SCD30 to take a look at why resetting is so unhappy on cPy?