adafruit/Adafruit_Blinka #401: ft232h: Does not implement SPI.read(..., write_value=0xff)

jepler, 2021-01-28:
This is at least a partial explanation of the problem reported in adafruit_circuitpython_sd: https://github.com/adafruit/Adafruit_CircuitPython_SD/issues/39

This is possibly the *only* SPI device that has this requirement.

After attempting to work around the limitation locally, I am able to get the sector count of a card:
```
import os
os.environ['BLINKA_FT232H'] = '1'

import board
import digitalio
import adafruit_sdcard

cs = digitalio.DigitalInOut(board.D4)
bus = board.SPI()
card = adafruit_sdcard.SDCard(bus, cs)
print(card.count())
```

However, my workaround is not suitable to include here (I implemented it in adafruit_sdcard).

This may affect other SPI implementations within adafruit_blinka.

Thanks to @ladyada who realized this could be the problem.

Gerd1986, 2021-01-28:
Thank You so much for checking on this! 
Can You Show me your workaround please?

Kind regards,

Gerd

jepler, 2021-01-28:
Likely also affected: nxp_lpc4330

closed 2021-02-08 by jepler