adafruit/Adafruit_Blinka #412: Neopixel not working on CM4

timonsku, 2021-02-05:
It complains about the hardware revision not being supported.
I tried fixing it in the underlying C lib (https://github.com/jgarff/rpi_ws281x) but when using the latest git version of that and patching it the Blinka library stops working entirely saying that the NeoPixel object doesn't exist. Was there maybe an API change?
It doesn't seem like anything but having the hardware ID in the detection code would prevent it from working.

ladyada, 2021-02-05:
nope we just call it directly - wheres the PR that it was fixed in ws28?

timonsku, 2021-02-05:
Oh I didn't create a PR I just hacked around locally. Created one now if you want to try it: https://github.com/jgarff/rpi_ws281x/pull/445
but even unmodified compiling and running the version from git threw that error that the NeoPixel object was missing from the library.

ladyada, 2021-02-05:
maybe not installed right? dunno what to say, we've been using ws28 for years now :/

timonsku, 2021-02-05:
It was working fine when using their own Python examples. I think its just a namespace issue somewhere. I'll investigate further.

timonsku, 2021-03-01:
I realize this issue should have probably went into the neopixel module?
Anyways I think the issue likely this: https://github.com/rpi-ws281x/rpi-ws281x-python/commit/60b034817e9dd2082b6a8eee0e108e214747bb5b#diff-b718f1354f7247312eca086d9a024afe5fa717ddea5adeddd6f12bcf945b2e8c

I can't see how the module https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel could have worked otherwise if rpi-ws281x-python didn't provide such a shim to have the same API as the CPY neopixel driver. I can't see any sort of wrapping going on otherwise. So this is likely a breaking change coming from the underlying python module. I guess the dependency just hadn't been updated in a long while and it installed the older working version by default?
I think this would either require reverting this change in `rpi-ws281x-python` or add a wrapper to `Adafruit_CircuitPython_NeoPixel` to work with the changed api.

@makermelissa @tannewt whats your position on that?

timonsku, 2021-03-01:
I also just realized that the python module is dependant on @pimoroni fork and not the original repo by jgarff.
This seems to be all maintained by @gadgetoid. This seems to also affect newer revisions of the Pi4 and not just the CM4 as that is also missing its hwid in the pimoroini fork.

https://github.com/pimoroni/rpi_ws281x

Gadgetoid, 2021-03-01:
This is largely due to me being in continued lockdown chaos and lacking the space for a proper Pi setup to vet and merge pulls for CM4 HWIDs. Will look into it ASAP. (also I don't have a CM4!)

timonsku, 2021-03-01:
thanks! Lemme know if I can help.

Gadgetoid, 2021-03-01:
The API - ie: the removal of "NeoPixel" - was changed at the request of @ladyada (https://github.com/rpi-ws281x/rpi-ws281x-python/issues/8) so that the ws281x library didn't trample on their Blinka Python library. Mentioned here - https://github.com/rpi-ws281x/rpi-ws281x-python/issues/10

I don't know quite if/how this relates to your issue, but I venture that might not be something I can fix.

As for the hardware IDs, I've merged them upstream and prepped the library for release... I can't build my usual (albeit not terribly comprehensive) glut of .whl files but I think Piwheels.org renders that step obsolete. A (test) source dist is available here- https://test.pypi.org/project/rpi-ws281x/4.2.6/

P.S. https://github.com/rpi-ws281x/rpi-ws281x-python/ is the source for the Python wrapper. I steered upstream `rpi_ws281x` away from including/maintaining Python bindings.

timonsku, 2021-03-01:
Huh ok, now I'm official confused how everything is supposed to go together 😄
I guess someone from Adafruit has to chime in here.

Thanks a lot for merging the hwids though. Looking forward to the next rpi-ws281x-python release.

timonsku, 2021-04-08:
After the whole dependency chain got update it is now all working!

closed 2021-04-08 by timonsku

makermelissa, 2021-04-08:
Excellent!