adafruit/Adafruit_Blinka #414: Error while installing this library

TinManAkshay, 2021-02-08:
Hi,

I am installing this library right now and I am getting this error shown below:

pi@raspberrypi:~ $ sudo pip3 install Adafruit_Blinka
Collecting Adafruit_Blinka
  Using cached https://www.piwheels.org/simple/adafruit-blinka/Adafruit_Blinka-5.13.1-py3-none-any.whl
Collecting Adafruit-PureIO>=1.1.7 (from Adafruit_Blinka)
  Using cached https://www.piwheels.org/simple/adafruit-pureio/Adafruit_PureIO-1.1.8-py3-none-any.whl
Requirement already satisfied: RPi.GPIO in /usr/lib/python3/dist-packages (from Adafruit_Blinka)
Collecting sysv-ipc (from Adafruit_Blinka)
  Using cached https://files.pythonhosted.org/packages/0c/d7/5d2f861155e9749f981e6c58f2a482d3ab458bf8c35ae24d4b4d5899ebf9/sysv_ipc-1.1.0.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-pysgk14t/sysv-ipc/setup.py", line 11, in <module>
        import prober
      File "/tmp/pip-build-pysgk14t/sysv-ipc/prober.py", line 137
        d["SYSV_IPC_VERSION"] = f'"{version}"'
                                             ^
    SyntaxError: invalid syntax
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-pysgk14t/sysv-ipc/

makermelissa, 2021-02-08:
What version of Python do you have? You can do `python3 --version`.

TinManAkshay, 2021-02-08:
It says Python 3.5.3

makermelissa, 2021-02-08:
Ok, you will need at least Python 3.6. Please do a `sudo apt update && sudo apt upgrade`.

makermelissa, 2021-02-08:
Gonna reopen this and change the nature of this to automatically inform user of Python version.

wolfmanjm, 2021-03-15:
FYI this error is only caused by the latest release of sysv-ipc package. As many rpi systems still use stretch and that uses python 3.5, rather than having to upgrade to buster tp get python 3.6, it would be much easier to simply change the requirement in Blinka for sysv-ipc to use v1.0.1 instead of v1.1.0, this allows the version of blinka on pypi to install ok on an rpi that still runs stretch.
Asking people to upgrade to python 3.6 or better on a pi zero for isntance is a non trivial ask. (and takes forever even if they can figure out how to do it).
For anyone running into this problem as of today you can simply do ```sudo pip3 install sysv_ipc==1.0.1``` then ```sudo pip3 install Adafruit-Blinka``` (However if blinka gets updated on pypi from 5.13.1 this workaround may stop working).

makermelissa, 2021-03-15:
Hi, if I recall correctly, the reason we changed to requiring users to run Python 3.6 was because CircuitPython itself was now targeting Python 3.6 language compatibility and we wanted to keep these in sync. 

@tannewt, you originally requested updating to Blinka to use Python 3.6, does that sound correct or was there another reason that I'm not remembering?

wolfmanjm, 2021-03-15:
3.6 is fine, so long as the user has upgraded to the latest version of raspberrypi raspbian (which is buster and uses python 3.7). I would suggest in the tutorials (which I think a lot of beginners will read as they are excellent), that it be spelled out that the latest raspbian is needed and not the older one based on stretch. An alternative is to have a link to a tutorial on how to install a newer version of Python on raspbian/stretch, updating the system python is probably not a great idea, but there are plenty of alternatives, I particularly like pyenv, but on a pizero it will take a long time to build a new version of python.