adafruit/Adafruit_Blinka #415: Blinka dumps modules in site-packages

thijstriemstra, 2021-02-09:
I installed Adafruit blinka and it's incorrectly installing itself into `site-packages`.

```console
$ ls -l ~/.virtualenvs/foo/lib/python3.7/site-packages/
drwxrwxr-x  7 thijs thijs   4096 jan  2 23:30 absl
drwxrwxr-x  2 thijs thijs   4096 jan  2 23:30 absl_py-0.11.0.dist-info
drwxrwxr-x  6 thijs thijs   4096 feb  9 19:04 adafruit_blinka
drwxrwxr-x  2 thijs thijs   4096 feb  9 19:04 Adafruit_Blinka-6.2.0.dist-info
drwxrwxr-x  4 thijs thijs   4096 jan 13 15:39 adafruit_platformdetect
drwxrwxr-x  2 thijs thijs   4096 jan  6 14:37 Adafruit_PlatformDetect-2.24.0.dist-info
drwxrwxr-x  3 thijs thijs   4096 jan  4 16:06 Adafruit_PureIO
drwxrwxr-x  2 thijs thijs   4096 jan  4 16:06 Adafruit_PureIO-1.1.8.dist-info
-rw-rw-r--  1 thijs thijs   1099 feb  9 19:04 analogio.py
drwxrwxr-x  5 thijs thijs   4096 jan  2 23:30 aniso8601
drwxrwxr-x  2 thijs thijs   4096 jan  2 23:30 aniso8601-8.1.0.dist-info
drwxrwxr-x  5 thijs thijs   4096 jan  2 23:30 aocr
drwxrwxr-x  2 thijs thijs   4096 jan  2 23:30 aocr-0.7.6.dist-info
drwxrwxr-x  3 thijs thijs   4096 jan  2 23:30 apipkg
drwxrwxr-x  2 thijs thijs   4096 jan  2 23:30 apipkg-1.5.dist-info
drwxrwxr-x  2 thijs thijs   4096 dec  6 15:27 appdirs-1.4.4.dist-info
-rw-rw-r--  1 thijs thijs  24720 dec  6 15:27 appdirs.py
drwxrwxr-x  3 thijs thijs   4096 jan  2 23:30 astor
drwxrwxr-x  2 thijs thijs   4096 jan  2 23:30 astor-0.8.1.dist-info
drwxrwxr-x  3 thijs thijs   4096 jan  3 03:59 astunparse
drwxrwxr-x  2 thijs thijs   4096 jan  3 03:59 astunparse-1.6.3.dist-info
drwxrwxr-x  3 thijs thijs   4096 jan  2 23:30 attr
drwxrwxr-x  2 thijs thijs   4096 jan  2 23:30 attrs-20.3.0.dist-info
-rw-rw-r--  1 thijs thijs   4422 feb  9 19:04 bitbangio.py
-rw-rw-r--  1 thijs thijs   7632 feb  9 19:04 board.py
-rw-rw-r--  1 thijs thijs  17862 feb  9 19:04 busio.py
drwxrwxr-x  2 thijs thijs   4096 jan  2 23:30 cached_property-1.5.2.dist-info
-rw-rw-r--  1 thijs thijs   4324 jan  2 23:30 cached_property.py
drwxrwxr-x  3 thijs thijs   4096 jan  3 03:58 cachetools
drwxrwxr-x  2 thijs thijs   4096 jan  3 03:58 cachetools-4.2.0.dist-info
drwxrwxr-x  4 thijs thijs   4096 feb  9 19:04 captcha22
drwxrwxr-x  2 thijs thijs   4096 feb  9 19:04 captcha22-1.0.6.dist-info
drwxrwxr-x  3 thijs thijs   4096 dec  6 16:42 certifi
drwxrwxr-x  2 thijs thijs   4096 dec  6 16:42 certifi-2020.12.5.dist-info
-rw-rw-r--  1 thijs thijs   8954 jan  6 14:36 CHANGES.md
# etc ...
```

As you can see it polluted the directory with all kinda of blinka modules, `analogio.py`, `board.py`, even `CHANGES.md` etc..

My guess is this setup is not working correctly: https://github.com/adafruit/Adafruit_Blinka/blob/master/setup.py#L45

ladyada, 2021-02-09:
thats normal, no? you want to be able to `import board` directly from Python

thijstriemstra, 2021-02-09:
Hmm I guess you're right, but at the same time there's a adafruit_blinka package, I would expect it to end up in there.

Reason I investigated this is because I'm getting this error after upgrading to latest blinka:

```
feb 09 19:16:12 snowblower python[45492]: Traceback (most recent call last):
feb 09 19:16:12 snowblower python[45492]:   File "src/python/hardware_ctl.py", line 20, in <module>
feb 09 19:16:12 snowblower python[45492]:     import board
feb 09 19:16:12 snowblower python[45492]:   File "/home/thijs/.virtualenvs/foo/lib/python3.7/site-packages/board.py", line 77, in <module>
feb 09 19:16:12 snowblower python[45492]:     elif board_id == ap_board.BEAGLEBONE_AI:
feb 09 19:16:12 snowblower python[45492]: AttributeError: module 'adafruit_platformdetect.constants.boards' has no attribute 'BEAGLEBONE_AI'
```

thijstriemstra, 2021-02-09:
Besides the traceback the bug here is the inclusion of CHANGES.md in site-packages. I'll see if I can make a PR for that.

ladyada, 2021-02-09:
make sure to upgrade _all_ your python packages including platform detect

thijstriemstra, 2021-02-09:
I ran:

```
pip install -U Adafruit-Blinka
```

again and:

```
Requirement already satisfied: Adafruit-PlatformDetect>=2.18.1 in /home/thijs/.virtualenvs/foo/lib/python3.7/site-packages (from Adafruit-Blinka) (2.24.0)
```

But 3.1.0 is the latest so you probably want to `Adafruit-PlatformDetect>=3.1.0` instead?

FoamyGuy, 2021-02-09:
I believe this is normal, and it does have to work this way in order for the same import code to get used on the microcontrollers vs. on Blinka compatibility layer. If all of those things were inside a module then the import statements would need to be changed to reflect that in pretty much any code ever written for Blinka.

thijstriemstra, 2021-02-09:
agreed @FoamyGuy, I forgot about this important feature.

thijstriemstra, 2021-02-09:
> But 3.1.0 is the latest so you probably want to Adafruit-PlatformDetect>=3.1.0 instead?

@ladyada see #416

thijstriemstra, 2021-02-09:
> Besides the traceback the bug here is the inclusion of CHANGES.md in site-packages. I'll see if I can make a PR for that.

The CHANGES.md wasn't a file from Blinka, sorry for the noise.

PR #416 (2021-02-10): bump minimum version for Adafruit-PlatformDetect
fixes #415
closed 2021-02-09 by ladyada