adafruit/Adafruit_Blinka #420: adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein fails in 64bit environment?

p3ck, 2021-02-22:
I'm a little perplexed on what is really wrong.  I'm trying to run this in a 64bit environment so that may be why things are failing. 

```
[root@rpi-living ~]# python3 dht_simpletest.py 
Traceback (most recent call last):
  File "dht_simpletest.py", line 9, in <module>
    dhtDevice = adafruit_dht.DHT22(board.D4)
  File "/usr/local/lib/python3.6/site-packages/adafruit_dht.py", line 274, in __init__
    super().__init__(False, pin, 1000, use_pulseio)
  File "/usr/local/lib/python3.6/site-packages/adafruit_dht.py", line 56, in __init__
    self.pulse_in = PulseIn(self._pin, 81, True)
  File "/usr/local/lib/python3.6/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py", line 67, in __init__
    self._process = subprocess.Popen(cmd)
  File "/usr/lib64/python3.6/subprocess.py", line 729, in __init__
    restore_signals, start_new_session)
  File "/usr/lib64/python3.6/subprocess.py", line 1364, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.6/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein': '/usr/local/lib/python3.6/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein'
```

uname -a
Linux rpi-living 5.4.88-v8.1.el8 #1 SMP PREEMPT Sun Jan 10 15:08:31 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux

more /etc/centos-release
CentOS Linux release 8.3.2011

I can get temp readings if I disable pulse:
```
[root@rpi-living ~]# python3 dht_simpletest.py 
Temp: 74.3 F / 23.5 C    Humidity: 30.2% 
Temp: 74.3 F / 23.5 C    Humidity: 30.2%
```

[root@rpi-living ~]# ldd /usr/local/lib/python3.6/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein
	not a dynamic executable
[root@rpi-living ~]# file /usr/local/lib/python3.6/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein
/usr/local/lib/python3.6/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 3.2.0, BuildID[sha1]=356bb870cf3f5a35b0862d12a708b782a9c365f6, not stripped

Any ideas?

makermelissa, 2021-02-22:
Yeah. I'm pretty sure the libgpiod_pulsein file is a binary that was compiled against the 32-bit environment. I can't remember the steps to compile, but take a look at #210, which I think is related. I think if you were to compile yourself and replace the file in your local file system, that would fix the issue for you. If that works, a long term solution may be to have 32 and 64 bit versions and have Blinka detect and use the appropriate one.

p3ck, 2021-02-22:
That worked like a charm!

```
git clone https://github.com/adafruit/libgpiod_pulsein.git
cd libgpiod_pulsein/src
make
cp libgpiod_pulsein /usr/local/lib/python3.6/site-packages/adafruit_blinka/microcontroller/bcm283x/pulseio/libgpiod_pulsein
```

It works fine now.

PR #506 (2021-10-05): Fix libgpiod for 64-bit Pi by detecting and loading appropriate lib
Fixes #420. Compiled libgpiod_pulsein on a 64-bit Raspberry Pi OS and added some detection code.
closed 2021-10-05 by kattni