adafruit/Adafruit_Blinka #423: FT232H: Pin should raise NotImplementedError if pull is not None

thomas6g, 2021-02-28:
Class Pin (in pin.py) of FT232H should raise _NotImplementedError_ if pull is not None to stay compatible. Additionally PULL_NONE, PULL_UP and PULL_DOWN should be defined. 

I'm using a DHT 22 with Adafruit Circuit Python DHT lib. The lib tests whether internal pull ups are supported. The test breaks without the mentioned changes.

ladyada, 2021-02-28:
and the DHT works through an FT232h?

thomas6g, 2021-02-28:
To be honest I don't know. It's my first try with the FTDI and the DHT was the first sensor I found in my pocket.

Unfortunately, I have only one DHT and it is not reacting at all. I checked with my scope but my sensor isn't responding to the start signal of the host. I feared that the timing might be a problem but I expected at least some reaction.
I just ordered another sample. I'll test again once it arrives. 

However, as I was playing around I thought I should report and fix #423.

ladyada, 2021-02-28:
ok - i really really do not think it will work, DHTs are terrible please just use an I2C sensor

thomas6g, 2021-02-28:
Sure - I ordered two BMP390L with some other stuff from Adafruit, yesterday.

I'm keen to start with my sensor project. May take a while until it reaches Germany...

thomas6g, 2021-02-28:
Will you accept the pull request?

I think we should fix the issue independent of the DHT theme.

ladyada, 2021-02-28:
yah we can look at the PR next week!

thomas6g, 2021-02-28:
Ok.

commit 0ee27bee (2021-03-01): Merge pull request #424 from thomas6g/thomas6g-tfdi232h-issue-423

Fix #423 FT232H Pin returns NotImplementedError
closed 2021-03-01 by ladyada