adafruit/Adafruit_Blinka #430: "import board" error in OPi zero plus 2 h5

meisamkassaeian, 2021-03-09:
when i import the "board" in my script i get this error:
The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.8/dist-packages/board.py", line 105, in <module>
    from adafruit_blinka.board.orangepi.orangepizeroplus2h5 import *
  File "/usr/local/lib/python3.8/dist-packages/adafruit_blinka/board/orangepi/orangepizeroplus2h5.py", line 3, in <module>
    from adafruit_blinka.microcontroller.allwinner.h5 import pin
  File "/usr/local/lib/python3.8/dist-packages/adafruit_blinka/microcontroller/allwinner/h5/pin.py", line 2, in <module>
    from adafruit_blinka.microcontroller.generic_linux.libgpiod_pin import Pin
  File "/usr/local/lib/python3.8/dist-packages/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py", line 5, in <module>
    raise ImportError(
ImportError: libgpiod Python bindings not found, please install and try again! See https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/master/libgpiod.sh

meisamkassaeian, 2021-03-09:
i want to use this for reading data from mpu6050 from i2c.

makermelissa, 2021-03-09:
You'll need to install the libgpiod Python bindings first. See https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/master/libgpiod.sh/

closed 2021-03-09 by makermelissa