adafruit/Adafruit_Blinka #435: Make sure to kill libgpiod threads

caternuson, 2021-03-18:
Not sure what a good answer to this is, or if it's even something that should be required of Blinka. But starting here.

This issue is a good example of what can happen:
https://github.com/adafruit/Adafruit_CircuitPython_DHT/issues/63
and more info in a forum related post here:
https://forums.adafruit.com/viewtopic.php?f=60&t=175209#p859042
which I'm copying here:

I think the general issue here is that these codes are all infinite loops that are being aborted using <CTRL><C>. As a result, the DHT exit() never gets called:
https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/ceb883d09587612d2f96037f54c93805e4175db3/adafruit_dht.py#L59
and so the pulseio deinit() does not get called:
https://github.com/adafruit/Adafruit_Blinka/blob/bc28caa0fe210741bda9586a134adad3794bda5e/src/adafruit_blinka/microcontroller/bcm283x/pulseio/PulseIn.py#L99
and so the background process handling the pulsein does not get terminated. So when the code is run a second time, that still running process still has the pin taken.

Further, the current example has some try/except logic:
https://github.com/adafruit/Adafruit_CircuitPython_DHT/blob/master/examples/dht_simpletest.py
but this will not catch a <CTRL><C> event since KeyboardInterrupt inherits from BaseException, not Exception:
https://docs.python.org/3/library/exceptions.html#KeyboardInterrupt

caternuson, 2021-03-18:
As suggested by @sommersoft , `atexit` may be an option.
https://docs.python.org/3/library/atexit.html#module-atexit

Quick test looks promising:
```python
import time
import atexit

@atexit.register
def cleanup():
    print("cleanup")

count = 0
while True:
    print("count =", count)
    count += 1
    time.sleep(1)
```

And run it and abort with `<CTRL><C>`:
```
~$ python3 exit_test.py 
count = 0
count = 1
count = 2
count = 3
count = 4
count = 5
^CTraceback (most recent call last):
  File "exit_test.py", line 12, in <module>
    time.sleep(1)
KeyboardInterrupt
cleanup

~$ 
```

jposada202020, 2021-03-18:
I used to use atexit inside classes, when I reall wnat to enforce it
```python
        atexit.register(self.cleanup)
```

caternuson, 2021-03-19:
This may be a Thonny specific quirk. I was unable to reproduce the issue running things from command line. It appears libgpiod already has some hooks to catch this:
https://forums.adafruit.com/viewtopic.php?f=60&t=175209#p862066

closed 2026-04-22 by makermelissa