adafruit/Adafruit_Blinka #439: Support for the original Beaglebone

Cellgalvano, 2021-04-05:
Hi,

I created a project using Blinka for the PocketBeagle and everything was working fine out of the box! Now I'd like to port it over to a few older original white Beaglebones but the board is not supported.

As far as I understand the SoC definition of the original Beaglebone is (mostly?) equal to the Beaglebone Black and therefore one could simply add another "else if" in boards.py for the board id "BEAGLEBONE" to support the older boards?

Thank you!

ladyada, 2021-04-05:
seems like it should work - we don't have an older 'Bone on hand, so if y'ad like to add support please send in a PR :)

pdp7, 2021-04-05:
@Cellgalvano it should work okay, just need to add BEAGLEBONE to:
https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/59edca95e6324678cf8ebef9f0356a30c7943369/adafruit_platformdetect/constants/boards.py#L291

The EEPROM ID on your BeagleBone should match one of these:
https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/master/adafruit_platformdetect/constants/boards.py#L291

pdp7, 2021-04-05:
@Cellgalvano please take a look at the proposed change in PR #440.  I don't have an original BeagleBone with me right now to test, but I think that should be what is needed.

Cellgalvano, 2021-04-06:
Hi,

thank you very much for the super fast response!
That is indeed the change I hoped for and for me it seems to work just fine. My board_id is set to 'BEAGLEBONE' as my Board is an original A6A Beaglebone and the detector got it right.

Best regards
Felix

pdp7, 2021-04-06:
@Cellgalvano good to hear.  is it ok to close the issue?

closed 2021-04-06 by Cellgalvano