adafruit/Adafruit_Blinka #442: Board not supported - BBB

zmoothrc, 2021-04-13:
Trying to revive my trusty ol'BBB, board not identified.

# python detect.py
Chip id:  AM33XX
Board id:  None

Is this a DragonBoard 410c? False
Is this a Pi 3B+? False
Is this a Pi 4B? False
Is this a 40-pin Raspberry Pi? False
Is this a Raspberry Pi Compute Module? False
Is this a BeagleBone Board? False
Is this a Giant Board? False
Is this a Coral Dev Board? False
Is this a Coral Dev Board Mini? False
Is this a SiFive Unleashed?  False
Is this a PYNQ Board? False
Is this a Rock Pi board? False
Is this a NanoPi board? False
Is this a Clockwork Pi board? False
Is this an embedded Linux system? False
Is this a generic Linux PC? False
Is this a UDOO Bolt? False
Is this an ASUS Tinker Board? False
Is this an STM32MP1 Board? False
Is this an OS environment variable special case? False

# cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 2 (v7l)
BogoMIPS	: 995.32
Features	: half thumb fastmult vfp edsp thumbee neon vfpv3 tls vfpd32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x3
CPU part	: 0xc08
CPU revision	: 2

Hardware	: Generic AM33XX (Flattened Device Tree)
Revision	: 0000
Serial		: 0000000000000000

pdp7, 2021-04-14:
I attempted to add support in PR #440 with commit https://github.com/adafruit/Adafruit_Blinka/pull/440/commits/c4248225afa49bc033970808f3c542fb5209de20

@makermelissa what is the best way to tell which version of Blinka and PlatfromDetect that someone is running?

makermelissa, 2021-04-14:
Run `pip3 show adafruit-blinka` which will also show you where it's installed.