adafruit/Adafruit_Blinka #445: Verify whether Blinka is working with MicroPython and fix if it is not

dhalbert, 2021-04-17:


jposada202020, 2021-04-17:
My experience, I tried last year to use it, not very successful. to be fair at that time I was not familiar with CircuitPython. My use case was to use i2c sensors, in particular BH1750 and AHT20. Problem was with the try_lock() function and read_from_into. At the end I had more experience with MP I ported the libraries to MicroPython. 
Let me know if something is needed I can re-try again and do some tests.

Red-M, 2021-04-18:
Doesn't work as this bug adafruit/Adafruit_Python_PlatformDetect#144 doesn't detect the rpi pico.

makermelissa, 2021-05-06:
The biggest issue I'm running into with getting this working on the PyBoard so far is that Blinka has grown quite a bit while the PyBoard's flash has not.

makermelissa, 2021-05-06:
Ha, I just realized that's an SD card slot on the board.

makermelissa, 2021-05-06:
Ok, I'm having some success now. Steps I did:
1. Updated MicroPython to latest stable (1.15)
2. Inserted a 16GB formatted MicroSD card
3. Reset the board
4. Copied the `adafruit_platformdetect` folder from the Adafruit_Python_PlatformDetect repo to the SD card (it shows up as the drive)
5. Copied everything inside the src folder of the `Adafruit_Blinka` repo
6. Soft reset for good measure
7. Tested with `import os` and `os.listdir()`
8. Used `import board` and `dir(board)` without errors.

makermelissa, 2021-05-06:
Here's the results for other modules:
Busio
```
>>> import busio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "busio.py", line 17, in <module>
  File "adafruit_blinka/agnostic/__init__.py", line 18, in <module>
  File "adafruit_platformdetect/chip.py", line 69, in id
AttributeError: 'module' object has no attribute 'environ'
```
Digitalio
```
>>> import digitalio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "digitalio.py", line 10, in <module>
ImportError: no module named 'adafruit_blinka.agnostic.board_id'
```
Analogio
```
>>> import analogio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "analogio.py", line 16, in <module>
  File "adafruit_platformdetect/board.py", line 560, in microchip_mcp2221
  File "adafruit_platformdetect/board.py", line 67, in id
AttributeError: 'module' object has no attribute 'environ'
```
Bitbangio
```
>>> import bitbangio
```
Pulseio
```
>>> import pulseio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "pulseio.py", line 16, in <module>
  File "adafruit_platformdetect/board.py", line 428, in any_raspberry_pi
  File "adafruit_platformdetect/board.py", line 197, in _pi_rev_code
  File "adafruit_platformdetect/chip.py", line 69, in id
AttributeError: 'module' object has no attribute 'environ'
```
Pwmio
```
>>> import pwmio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "pwmio.py", line 16, in <module>
  File "adafruit_platformdetect/board.py", line 428, in any_raspberry_pi
  File "adafruit_platformdetect/board.py", line 197, in _pi_rev_code
  File "adafruit_platformdetect/chip.py", line 69, in id
AttributeError: 'module' object has no attribute 'environ'
```

So it looks like there are a few things to fix.

makermelissa, 2021-05-06:
I realized I wasn't testing with the latest PlatformDetect. With the latest I get something similar for import board:
```
>>> import board
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "board.py", line 33, in <module>
  File "adafruit_blinka/agnostic/__init__.py", line 18, in <module>
  File "adafruit_platformdetect/chip.py", line 69, in id
AttributeError: 'module' object has no attribute 'environ'
```

makermelissa, 2021-05-06:
Import errors fixed by https://github.com/adafruit/Adafruit_Python_PlatformDetect/pull/161. Now to do further testing.

makermelissa, 2021-05-06:
@Red-M, that's a separate issue. I created a new issue at https://github.com/adafruit/Adafruit_Python_PlatformDetect/issues/162.

makermelissa, 2021-05-07:
Digitalio works great now. I'll test I2C, SPI, and UART tomorrow.

makermelissa, 2021-05-07:
Ok, so far I got I2C and SPI working. PR coming soonish.

makermelissa, 2021-05-07:
UART is working. Tested everything on the Raspberry Pi Pico. Just testing this all on the PyBoard before submitting a PR.

PR #461 (2021-05-09): Fix MicroPython and busio
Fixes #445. Tested with PyBoard and Raspberry Pi Pico.
closed 2021-05-09 by makermelissa