adafruit/Adafruit_Blinka #446: Add support for https://github.com/execuc/u2if

ladyada, 2021-04-19:
would be a fun Blinka addition! could be a good partner for a qtpy 2040

makermelissa, 2021-04-22:
I think @caternuson was working on this, so I assigned. If not we can remove.

caternuson, 2021-04-22:
yep. prolly pr in a couple days.

caternuson, 2021-04-22:
@ladyada For anything other than Pico, would likely need to do a custom firmware build, since it picks specific pins for I2C/SPI:
https://github.com/execuc/u2if#u2if-pinout

ladyada, 2021-04-22:
yah understood, lets start with pico, then do QT Py or Feather 2040

2231puppy, 2021-05-04:
Just tried it out with [this tutorial](https://learn.adafruit.com/circuitpython-libraries-on-any-computer-with-raspberry-pi-pico/overview). Seems to work fine(mostly), but I can't get SPI to work on an ST7789 display(Adafruit PID[4313](https://www.adafruit.com/product/4313)). Any ideas? Complete error message:

`Traceback (most recent call last):
  File "blinkatest.py", line 27, in <module>
    disp.image(image)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/adafruit_rgb_display/rgb.py", line 200, in image
    self._block(x, y, x + imwidth - 1, y + imheight - 1, pixels)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/adafruit_rgb_display/rgb.py", line 145, in _block
    self.write(self._RAM_WRITE, data)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/adafruit_rgb_display/rgb.py", line 297, in write
    spi.write(data)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/busio.py", line 339, in write
    return self._spi.write(buf, start, end)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/adafruit_blinka/microcontroller/pico_u2if/spi.py", line 49, in write
    pico_u2if.spi_write(buf, start=start, end=end)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/adafruit_blinka/microcontroller/pico_u2if/pico_u2if.py", line 324, in spi_write
    print(bytes([write_cmd, chunk]) + buffer[start : (start + chunk)], True)
TypeError: can't concat list to bytes`

2231puppy, 2021-05-04:
Forgot to post the code 🤦🏻 [It's here](https://pastebin.com/qPwquewW)

caternuson, 2021-05-04:
Could be that something is broken with the chunkification. I did test with a SPI TFT though. @2231puppy Can you please open this as a new issue?

I think this issue can actually be closed? Support was added with #449. For any follow on issues, we'll open new ones.

ladyada, 2021-05-04:
thanks micah!

closed 2021-05-04 by ladyada

caternuson, 2021-05-04:
Oops. Or maybe not? Support was added for Pico, but not for Feather, Itsy, Qt Py. So technically, not fully resolved. But can just open new issue(s) for those?

ladyada, 2021-05-04:
yah we will do that later!