adafruit/Adafruit_Blinka #447: Simplify board and processor importing

makermelissa, 2021-04-19:
At this point, there are a lot of board that are handled in Blinka and the original way of having a huge if/elif block of boards is starting to get a bit silly. I'm wondering if we can do something more clever like keeping a data file (maybe json or some other format) that holds the detected board name, the import path, supported features, etc and just do a lookup in there.

One advantage of this is it would give a central place to add new boards and then all the scripts could just look up in there. Like let's say we add board xyz and it has GPIO and I2C working, but nothing else. So then for instance, the analogio script would see that the analogio feature isn't supported and report back that the feature isn't implemented on board xyz.

Anyway, this was a quick idea to reduce the complexity of the code and it will probably need to be altered during implementation.

ladyada, 2021-05-01:
plz check with me before you implement anything

makermelissa, 2022-11-28:
This came up again in my head while updating the detect.py file in PlatformDetect, so I thought I'd add more detail before I forget again. 

Originally I think the idea was basically to have a LUT that just associated the detected Chip to the microcontroller import and the detected board to the board import, but have it in JSON format rather than a huge if/else block. The purpose would be to have a single place to have a lookup similar to the idea behind detect. Originally I was also thinking It could maybe have features for each board, but upon further thought, I don't think that's an improvement at all.

I thought of a couple of alternative ideas like a class that imports once or even having board.py be the main place to have the import tree. However, the more I think about it, I'm really not sure if any of these ideas is better than the current implementation and perhaps a little experimentation is in order when I have a bit more time to focus on Blinka as a whole. Maybe a generator that stops parsing through imports once it finds the correct one? Anyways, this was mostly just to provide a bit more clarity to "future me".

makermelissa, 2023-03-27:
I wonder if something like this could be leveraged to create a blinka.json file similar to https://github.com/adafruit/circuitpython-org/tree/main/_data to make maintaining a list of the Blinka boards on circuitpython.org much easier.

anguselheimStudios, 2023-05-06:
I guess this applies here more than #609, but here is what I was my immediate thought when I saw the huge if/elif blocks in platform_detect:
```
def platform_detect_example():
    try:
        with open(
            "/proc/device-tree/compatible", "r", encoding="utf-8"
        ) as model_file:
            dt_compatible = model_file.read().split(',')
        print('The board definition for this SBC should be named:')
        print(f'{dt_compatible[0]}_{dt_compatible[1]}.py')
    except FileNotFoundError:
        pass
    return None
```
If my understanding is correct, on a RPi 4b you should get "raspberrypi_4-model-b.py"
With all the definitions labeled correctly you could then do something like `exec(f'import board_definition.{dt_compatible[0]}_{dt_compatible[1]}'}` or whatever.

makermelissa, 2026-02-02:
Just to see if it could be done, I came up with https://github.com/makermelissa/Adafruit_Blinka/blob/jsonify/src/board.py and https://github.com/makermelissa/Adafruit_Blinka/blob/jsonify/src/board_imports.json. The combined line total is about half of what it is currently. We could add some more data in there as well like the board name. Something similar could be done for the microprocessors (which are in at least 2 places).

I think this would make it easier to maintain and has the additional benefit of making it easier to track the boards that are supported by Blinka.

@ladyada, what do you think?