adafruit/Adafruit_Blinka #448: Support for Ubuntu Server 20.04.2 LTS

incomingstorm, 2021-04-21:
It would be great if support for Ubuntu Server 20.04.2 LTS for Raspberry Pi would be added.

makermelissa, 2021-05-05:
Hi, have you tried it? I had heard it was working on Ubuntu Desktop despite not officially being supported. If it doesn't work, what is the error?

makermelissa, 2021-05-06:
Closing for now because we don't officially support ubuntu on the Raspberry Pi. Feel free. to submit a PR.

closed 2021-05-06 by makermelissa

2bndy5, 2021-05-25:
I just ran into this problem myself. It would seem that RPi.GPIO cannot install to ubuntu on RPi (possibly because there isn't a pre-built wheel for `aarch64-linux` or RPi.GPIO is only meant to work on ARM-vX). I have posted the error output from `sudo pip3 install adafruit_blinka` below. I had to use `sudo` to abate the multiple "WARNING: Keyring is skipped due to an exception: Failed to unlock the collection!" when not using `sudo`

```shell
brendan@rpi4-ubuntu64:~$ sudo pip3 install adafruit_blinka
Processing /root/.cache/pip/wheels/ed/6e/ab/bee6cf4b9f950e1505327677e23373e436d2b9c635e9a1d067/Adafruit_Blinka-6.10.0-py3-none-any.whl
Processing /root/.cache/pip/wheels/1c/9e/87/e4bda2c653281b658d27116fdf208372cc333b7da3bc9e813d/rpi_ws281x-4.2.6-cp38-cp38-linux_aarch64.whl
Requirement already satisfied: Adafruit-PlatformDetect>=3.13.0 in /usr/local/lib/python3.8/dist-packages (from adafruit_blinka) (3.13.3)
Collecting pyftdi>=0.40.0
  Using cached pyftdi-0.53.1-py3-none-any.whl (141 kB)
Collecting RPi.GPIO
  Using cached RPi.GPIO-0.7.0.tar.gz (30 kB)
Requirement already satisfied: Adafruit-PureIO>=1.1.7 in /usr/local/lib/python3.8/dist-packages (from adafruit_blinka) (1.1.8)
Processing /root/.cache/pip/wheels/a5/8d/f7/0d28f2382ea77b66b0a64eb1bc43b10eb07aa8bd5e7369d0b2/sysv_ipc-1.1.0-cp38-cp38-linux_aarch64.whl
Collecting pyserial>=3.0
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting pyusb>=1.0.0
  Using cached pyusb-1.1.1-py3-none-any.whl (58 kB)
Building wheels for collected packages: RPi.GPIO
  Building wheel for RPi.GPIO (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-lcb1zb5s/RPi.GPIO/setup.py'"'"'; __file__='"'"'/tmp/pip-install-lcb1zb5s/RPi.GPIO/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-_sd22xft
       cwd: /tmp/pip-install-lcb1zb5s/RPi.GPIO/
  Complete output (85 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-aarch64-3.8
  creating build/lib.linux-aarch64-3.8/RPi
  copying RPi/__init__.py -> build/lib.linux-aarch64-3.8/RPi
  creating build/lib.linux-aarch64-3.8/RPi/GPIO
  copying RPi/GPIO/__init__.py -> build/lib.linux-aarch64-3.8/RPi/GPIO
  running build_ext
  building 'RPi._GPIO' extension
  creating build/temp.linux-aarch64-3.8
  creating build/temp.linux-aarch64-3.8/source
  aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.8 -c source/py_gpio.c -o build/temp.linux-aarch64-3.8/source/py_gpio.o
  aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.8 -c source/c_gpio.c -o build/temp.linux-aarch64-3.8/source/c_gpio.o
  source/c_gpio.c: In function ‘setup’:
  source/c_gpio.c:130:9: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
    130 |     if ((uint32_t)gpio_mem % PAGE_SIZE)
        |         ^
  source/c_gpio.c:131:34: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
    131 |         gpio_mem += PAGE_SIZE - ((uint32_t)gpio_mem % PAGE_SIZE);
        |                                  ^
  aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.8 -c source/cpuinfo.c -o build/temp.linux-aarch64-3.8/source/cpuinfo.o
  source/cpuinfo.c: In function ‘get_rpi_info’:
  source/cpuinfo.c:139:28: warning: format ‘%llx’ expects argument of type ‘long long unsigned int *’, but argument 3 has type ‘uint64_t *’ {aka ‘long unsigned int *’} [-Wformat=]
    139 |       sscanf(revision, "%llx", &rev);
        |                         ~~~^   ~~~~
        |                            |   |
        |                            |   uint64_t * {aka long unsigned int *}
        |                            long long unsigned int *
        |                         %lx
  aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.8 -c source/event_gpio.c -o build/temp.linux-aarch64-3.8/source/event_gpio.o
  aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.8 -c source/soft_pwm.c -o build/temp.linux-aarch64-3.8/source/soft_pwm.o
[... cut ...] (__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-58941u0q/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python3.8/RPi.GPIO Check the logs for full command output.
```

I don't expect this get fixed (because it seems more like a problem with RPi.GPIO to me), but I thought I'd post the error output as lead for others passing-by this thread.