adafruit/Adafruit_Blinka #451: LED Display Matrix Rows Missing

schnipdip, 2021-05-01:
This issue needs to exist for the workaround:

#https://github.com/adafruit/Adafruit_Blinka/issues/150

**Issue:**

When I instantiate and utilize the #https://github.com/hzeller/rpi-rgb-led-matrix  50% of the display matrix isn't initialized. 

Code:
```
def matrix_options():
    '''
        Params:
            options     - Display Board connection options
            matrix      - Sets the options
        Returns:
            matrix, offscreen_canvas
    '''
    
    options = RGBMatrixOptions()

    options.rows = 16
    options.cols = 32
    options.gpio_slowdown = 4
    options.brightness = 100
    options.chain_length = 1
    options.parallel = 1
    options.row_address_type = 0
    options.pwm_bits = 11
    options.pwm_lsb_nanoseconds = 130
    options.led_rgb_sequence = 'RGB' 
    options.pixel_mapper_config = ''
    #options.panel_type = '' # -> supported: str('FM6126A')
    options.show_refresh_rate = 0
    #options.hardware_mapping = 'regular'

    matrix = RGBMatrix(options = options)
    
    offscreen_canvas = matrix.CreateFrameCanvas()

    return (matrix, offscreen_canvas)

graphics.DrawText(matrix, font, 2, 12, red, 'BACK')
```

**Fix:**
Removing root from the gpio group and executing the python script with sudo while the .service or main python file is running in the background.

```
RuntimeError: Not running on a RPi!
```

[image]

[image]

[image]

[image]

ladyada, 2021-05-01:
this issue doesnt make much sense here? the RGB matrix code doesnt rely on Blinka at all. maybe some other code does. if you have questions about rgb matrix that's a different project/repo

schnipdip, 2021-05-01:
> this issue doesnt make much sense here? the RGB matrix code doesnt rely on Blinka at all. maybe some other code does. if you have questions about rgb matrix that's a different project/repo

Hard to tell. The interpreter is throwing a bunch of blinka and digitalio errors. I opened the same issue in #https://github.com/hzeller/rpi-rgb-led-matrix/issues/1299

ladyada, 2021-05-01:
it sounds like you need help with a project, not a bug report. you can post in the adafruit forums with *minimal* examples that demonstate the issue!

closed 2021-05-01 by ladyada

schnipdip, 2021-05-01:
What? This is literally the definition of a bug... 

Bugs are not features... Just as a reminder.

ladyada, 2021-05-01:
you can open a bug report with a bug report, that will require a minimal test example and the text output of failure (please dont post photographs of a screen).

schnipdip, 2021-05-01:
It's because there is intention to show flow. How can I demo the physical LED lights to properly explain what is being experienced without photos?

If you want to ask questions that's fine or want more elaboration that's fine. But that doesn't justify lazy development.

ptorrone, 2021-05-01:
@schnipdip that's not a good or acceptable way to work together here if there are issues we can assist with and you want help with something. we are closing this issue and please work towards communicating with folks here better - "lazy development" etc. leave things like that out.

schnipdip, 2021-05-01:
> @schnipdip that's not a good or acceptable way to work together here if there are issues we can assist with and you want help with something. we are closing this issue and please work towards communicating with folks here better - "lazy development" etc. leave things like that out.

No. "it sounds like you need help with a project, not a bug report."