adafruit/Adafruit_Blinka #454: Can't use an ST7789 TFT with Pico U2IF

2231puppy, 2021-05-04:
Hello, I can't seem to get my TFT(linked [here](https://www.adafruit.com/product/4313)) working with a Pi Pico running U2IF. It always gives me this error:

```
File "blinkatest.py", line 55, in <module>
    disp.image(image)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/adafruit_rgb_display/rgb.py", line 200, in image
    self._block(x, y, x + imwidth - 1, y + imheight - 1, pixels)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/adafruit_rgb_display/rgb.py", line 145, in _block
    self.write(self._RAM_WRITE, data)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/adafruit_rgb_display/rgb.py", line 297, in write
    spi.write(data)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/busio.py", line 339, in write
    return self._spi.write(buf, start, end)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/adafruit_blinka/microcontroller/pico_u2if/spi.py", line 49, in write
    pico_u2if.spi_write(buf, start=start, end=end)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/adafruit_blinka/microcontroller/pico_u2if/pico_u2if.py", line 326, in spi_write
    bytes([write_cmd, chunk]) + buffer[start : (start + chunk)], True
TypeError: can't concat list to bytes
```

I've tinkered around for ~3hrs after posting this problem in #446 and being asked to put it in a new issue. Not wanting to immediately do that, I messed with the code for a while until I turned to y'all, the Python wizards. The problem seems to be that the PIL object can't be converted to bytes, but if you use the raw image data as the bytes, it complains about it not having the right attributes. Might be totally wrong about that; just my best guess.

Thanks!

makermelissa, 2021-05-04:
I think the issue is the buffer is being passed in as a list instead of a bytestring. 

It can be reproduced in Python by attempting to concatenate a list of bytes to a bytestring:
```
>>> buffer = [0x23, 0x03, 0x65, 0x67, 0x34]
>>> bytes([0x62, 10]) + buffer[1:3]
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can't concat list to bytes
```
A bytestring doesn't produce an error:
```
>>> buffer = b"\x23\x03\x65\x67\x34"
>>> bytes([0x62, 10]) + buffer[1:3]
b'b\n\x03e'
```
Encapsulating the buffer in `bytes()` seems to work for both data types:
```
>>> buffer = [0x23, 0x03, 0x65, 0x67, 0x34]
>>> bytes([0x62, 10]) + bytes(buffer[1:3])
b'b\n\x03e'
>>> buffer = b"\x23\x03\x65\x67\x34"
>>> bytes([0x62, 10]) + bytes(buffer[1:3])
b'b\n\x03e'
```
So changing https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/microcontroller/pico_u2if/pico_u2if.py#L325 from
`bytes([write_cmd, chunk]) + buffer[start : (start + chunk)], True` to `bytes([write_cmd, chunk]) + bytes(buffer[start : (start + chunk)]), True` should fix the issue. Want to test that on your end?

2231puppy, 2021-05-04:
Ok, I'm really confused because it just started working again without me changing anything. I'm trying to break it again so I can test your solution(🙃 )

2231puppy, 2021-05-04:
I figured it out. I switched to PyPy at some point to see if an alternate solution I had in mind would work(it didn't.) I kept using PyPy for the fun of it and get this: The Blinka line of code that was causing my problem in default Python 3 was fine in PyPy! I'm going to test your solution to see if it fixes the default version of Python.

2231puppy, 2021-05-04:
Just gave it a shot and it works like a charm with the change. Should I put in a PR?

ladyada, 2021-05-04:
@2231puppy yeah go for it !

caternuson, 2021-05-05:
FWIW, this was the TFT test I used which worked fine with the Pico u2if:
```python
import board
import busio
import digitalio
import adafruit_rgb_display.ili9341 as ili9341
from PIL import Image

cs = digitalio.DigitalInOut(board.GP7)
dc = digitalio.DigitalInOut(board.GP8)
rst = digitalio.DigitalInOut(board.GP9)
spi1 = busio.SPI(board.SCLK1, board.MOSI1, board.MISO1)

disp = ili9341.ILI9341(spi1, baudrate = 24000000, cs=cs, dc=dc, rst=rst)

img = Image.open("test.jpg")

disp.image(img)
```

The code you linked in the other thread is creating the image with this:
```python
image = Image.new("RGB", (width, height))
```
which must return something slightly different then `open()`.

Wonder if this would work by just doing some kind of conversion, or calling some helper function, on the `image` object returned by `new` before passing it to `disp.image()` ?

Otherwise, I guess we could coherence the buffer to be `bytes`, per your PR.

2231puppy, 2021-05-05:
@caternuson It doesn't seem to work when I change Image.new to Image.open. It still gives me the `can't concat list to bytes` error. Must be something with the driver library.

caternuson, 2021-05-05:
OK, weird. It doesn't work for me either now. I'm seeing same error trying to run that test code above. Hmmmm....wonder what happened? Pretty sure it used to work. Let me dig into this a little.

caternuson, 2021-05-05:
I think the real culprit might be here:
https://github.com/adafruit/Adafruit_CircuitPython_RGB_Display/blob/d63cae507b84a6e3b5d958d5b5d220fdca37b67f/adafruit_rgb_display/rgb.py#L191

@makermelissa Any chance you know the reasoning for that cast to list?

makermelissa, 2021-05-05:
Not sure @caternuson. Looking at the blame, it looks like it was part of this commit: https://github.com/adafruit/Adafruit_CircuitPython_RGB_Display/commit/4a18be93ed5a9044a5b976fe91970ae1941579d4. It would probably be safe to return change it so it returns bytes instead of a list.

caternuson, 2021-05-05:
Ah, good sleuthing! Thanks @makermelissa . And agree.

hey @2231puppy want to try and PR a fix for that? It'd be a new PR to the Adafruit_CircuitPython_RGB repo that changes that line to:
```python
            pixels = bytes(image_to_data(img))
```

2231puppy, 2021-05-06:
Done! PR [here](https://github.com/adafruit/Adafruit_CircuitPython_RGB_Display/pull/93)

2231puppy, 2021-05-06:
The PR has been merged and issue is fixed, closing.

closed 2021-05-06 by 2231puppy