adafruit/Adafruit_Blinka #462: Manjaro ARM Linux - "Setup can't determine the value of PAGE_SIZE on your system"

spiritedfang, 2021-05-09:
When I try running `sudo pip3 install adafruit-circuitpython-ssd1306`, I get the following that prevents me from continuing:

```
Installing collected packages: sysv-ipc, rpi-ws281x, RPi.GPIO, pyftdi, Adafruit-PureIO, Adafruit-PlatformDetect, Adafruit-Blinka, adafruit-circuitpython-framebuf, adafruit-circuitpython-busdevice, adafruit-circuitpython-ssd1306
    Running setup.py install for sysv-ipc ... error
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-52udnwi9/sysv-ipc_aa3679015dd04294b270941ded18131d/setup.py'"'"'; __file__='"'"'/tmp/pip-install-52udnwi9/sysv-ipc_aa3679015dd04294b270941ded18131d/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-cs_8w_9d/install-record.txt --single-version-externally-managed --compile --install-headers /usr/include/python3.9/sysv-ipc
         cwd: /tmp/pip-install-52udnwi9/sysv-ipc_aa3679015dd04294b270941ded18131d/
    Complete output (15 lines):
    ******************************************************************************
    * Setup can't determine the value of PAGE_SIZE on your system, so it will
    * default to 4096 which may not be correct.
    *
    * Please report this message and your operating system info to the package
    * maintainer listed in the README file.
    ******************************************************************************
    running install
    running build
    running build_ext
    building 'sysv_ipc' extension
    creating build
    creating build/temp.linux-aarch64-3.9
    gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -march=armv8-a -O3 -pipe -fno-plt -fno-semantic-interposition -march=armv8-a -O3 -pipe -fno-plt -march=armv8-a -O3 -pipe -fno-plt -fPIC -I/usr/include/python3.9 -c common.c -o build/temp.linux-aarch64-3.9/common.o
    error: command 'gcc' failed: No such file or directory
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-52udnwi9/sysv-ipc_aa3679015dd04294b270941ded18131d/setup.py'"'"'; __file__='"'"'/tmp/pip-install-52udnwi9/sysv-ipc_aa3679015dd04294b270941ded18131d/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-cs_8w_9d/install-record.txt --single-version-externally-managed --compile --install-headers /usr/include/python3.9/sysv-ipc Check the logs for full command output.
```

spiritedfang, 2021-05-09:
Just some more details:

- Device: Raspberry Pi 4 Model B
- OS: Manjaro Linux ARM x64

I tried Ubuntu Server ARM x64 instead just to see if it was a 64-bit thing, but that worked.

ladyada, 2021-05-09:
please try raspian

spiritedfang, 2021-05-09:
It works on Raspbian. I didn't realize this requires Raspbian, that's my bad.

closed 2021-05-09 by spiritedfang