adafruit/Adafruit_Blinka #466: Use I2C on unsupported linux boards

fgervais, 2021-05-10:
I'm sorry this is not an issue but I though that could be useful and didn't really know where to post it.

Since linux uses a generic way to access the i2c bus we can still use it with blinka even if the board in unsupported.

For example, here's what I did for the tmp117:

```python
import adafruit_tmp117
import time

from adafruit_blinka.microcontroller.generic_linux.i2c import I2C as LINUX_I2C
from busio import I2C as BUSIO_I2C

class I2C(BUSIO_I2C):
    def __init__(self):
        self._i2c = LINUX_I2C(0)

i2c = I2C()
tmp117 = adafruit_tmp117.TMP117(i2c)

while True:
    temp = tmp117.temperature
    print(f"Temperature: {temp}°C")

    time.sleep(5)
```

ladyada, 2021-05-10:
we need to have a board definition because many libraries need gpio access - or there may be multiple I2C ports. we'd perfer to fix board support!

fgervais, 2021-05-10:
I always feel flattered to receive a reply from the ladyada :) Is this a shared account for the company or something?

I agree adding board is definitely the way to go.

ladyada, 2021-05-10:
ok please make an issue for the new board, or follow our guide to add it
https://learn.adafruit.com/adding-a-single-board-computer-to-blinka
we help people every day add new boards, really its the way to go!

closed 2021-05-10 by ladyada