adafruit/Adafruit_Blinka #472: file system issues on Raspberry pi Pico

jerryneedell, 2021-05-20:
I went through the guide to install Blinka on a Raspberry Pi Pico and ran into a few issues.  I had some files already on the system and when I went to copy everything over, I started getting "management Errors" in Thonny. I think it was running out of space. The aadafruit_blinka src folder is over 1 Mbyte itself!  I finally got it to work by deleting everything and reinstalling. The bme280 example runs fine. But if I try to copy more files to the Pico I get the "management error" again. 
So,  just to test,  I went into the adafruit_blinka/board folder and remved everything except the raspberrypi folder and the pioc_u2if.py file  -- It still works!!(edited)

Perhaps it would be best to only load what is needed.

If I try to look at the file system with uos.statvfs("/") I still get odd results 
(4096, 4096, 352, -53, -53, -0, 0, 0, 0, 255)
but it is better than it was with all the the files loaded ( the -53 was -139) 
I'm confused as to what the negative numbers mean for free space....

jerryneedell, 2021-05-20:
I was able to also remove all but the rp2040 and pico_u2if folders from the adafruit_blinka/microcontroller folder -- that saves a lot of space!
THe bme280 example still runs!

makermelissa, 2021-05-20:
I was going to suggest that @jerryneedell. You can even remove the pico_u2if folder. Many of the boards can probably be deleted as well.

caternuson, 2021-05-20:
Just to make sure there's no confusion - you're trying to do this:
https://learn.adafruit.com/circuitpython-libraries-on-micropython-using-the-raspberry-pi-pico
not this:
https://learn.adafruit.com/circuitpython-libraries-on-any-computer-with-raspberry-pi-pico
?

jerryneedell, 2021-05-20:
@caternuson  Yes, the first one!
@makermelissa  yes, I have removed all boards except raspberry pi and all microcontrollers except rp2040.

jposada202020, 2021-05-20:
I had the same problems, my solutions were to remove some of the boards as makermelissa mentioned

jposada202020, 2021-05-20:
I ran into a different issue, @makermelissa I discussed in the discord dev channel let me know if you want me to make an issue for that.

makermelissa, 2021-05-20:
@jposada202020 yep I saw, but hadn't had a chance to reply. I'm also trying to work with the maintainer (maybe creator too) of micropip to make a minimal install easy.

@caternuson I mentioned your guide on the first page of my guide.

makermelissa, 2021-05-20:
@jposada202020 also, please do create an issue. I'm going to take a look at it right away. Thanks.

makermelissa, 2021-05-27:
I've updated the guide to suggest stripping out the unnecessary stuff for this setup.

closed 2021-05-27 by makermelissa