adafruit/Adafruit_Blinka #473: Using Blinka for with Micropython in a Raspberry Pi Pico

jposada202020, 2021-05-20:
I was trying to follow learning guide
https://learn.adafruit.com/circuitpython-libraries-on-micropython-using-the-raspberry-pi-pico

### Hardware
```python
MicroPython v1.15 on 2021-04-18; Raspberry Pi Pico with RP2040
```

However I Used a different Sensor
I used an **Adafruit ADT7410 I2C Temperature Sensor** with the latest Blinka 6.9.1 and plataform detect 3.13.1

### Results
While reading the sensor the following I got the following error message:
```python
Traceback (most recent call last):
  File "<stdin>", line 10, in <module>
  File "/lib/adafruit_adt7410.py", line 94, in __init__
  File "/lib/adafruit_adt7410.py", line 130, in _read_register
  File "/lib/adafruit_bus_device/i2c_device.py", line 125, in write_then_readinto
  File "/lib/busio.py", line 147, in writeto_then_readfrom
  File "/lib/adafruit_blinka/microcontroller/rp2040/i2c.py", line 55, in writeto_then_readfrom
AttributeError: 'I2C' object has no attribute 'writeto_then_readfrom'
```

makermelissa, 2021-05-20:
I'm working on this. I've added this on my board, but it's somewhat consistently returning the incorrect value for the ID. According to docs and code, it should return 0xC8, but I'm getting stuff like 0x90, 0x98, 0xA0, 0xA8, and 0xB0. The characters returned right before and after are consistently always 0x0A and 0x80. I've tried adding delays, but that didn't help.

I'll try different speeds to see if that makes a difference. I might need to hook up a saleae to see if it's actually returning those values. Also, it may be related to https://github.com/adafruit/Adafruit_CircuitPython_ADT7410/issues/4.

jposada202020, 2021-05-20:
Let me know if a need to test with another sensor, I just picked the first one in my box :(

makermelissa, 2021-05-20:
Ok, making progress. The read value matches on the Saleae, but the write value does not.

makermelissa, 2021-05-20:
It's actually a good one because it's failing and it's a problem I can reproduce.

makermelissa, 2021-05-20:
Oh wait, no. Both values match. This one is tricky.

makermelissa, 2021-05-20:
Sooo, I just did a read and write using command line pure MicroPython and it's also giving unexpected values. 

As an added complication, I could only find my ADT7410 featherwing and it has a ADXL343 which appears to also use I2C, though at a different address. It should work, but it's a potential factor. I should have a breakout somewhere, but couldn't locate it.

It seems that there's something that needs fixing on the MicroPython side of things. I did improve the writeto and readfrom_into functions by adding the stop values back in, so I'll get this buttoned up for now and it does fix the original error in this issue.

PR #475 (2021-05-20): Added more stuff to RP2040 I2C
Fixes #473. This also adds the stop parameters to readfrom_into and writeto. Verified that this still works with the bme280 example.
closed 2021-05-20 by makermelissa