adafruit/Adafruit_Blinka #478: ST7735R not working with PICO_U2IF (Raspberry PICO board)

cs0ckn, 2021-05-23:
When using the ST7735R display on the Mini TFT featherwing with PICO_U2IF on a Raspberry PICO I get error:-

Exception in thread Thread-1:
Traceback (most recent call last):
  File "I:\Python\lib\threading.py", line 954, in _bootstrap_inner
    self.run()
  File "I:\Python\lib\threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "I:\Python\lib\site-packages\displayio\display.py", line 248, in _refresh_loop
    self.refresh()
  File "I:\Python\lib\site-packages\displayio\display.py", line 244, in refresh
    self._refresh_display_area(area)
  File "I:\Python\lib\site-packages\displayio\display.py", line 288, in _refresh_display_area
    self._write(self._write_ram_command, pixels)
  File "I:\Python\lib\site-packages\displayio\display.py", line 204, in _write
    self._bus.send(False, data)
  File "I:\Python\lib\site-packages\displayio\fourwire.py", line 116, in send
    self._spi.write(data)
  File "I:\Python\lib\site-packages\busio.py", line 341, in write
    return self._spi.write(buf, start, end)
  File "I:\Python\lib\site-packages\adafruit_blinka\microcontroller\pico_u2if\spi.py", line 49, in write
    pico_u2if.spi_write(buf, start=start, end=end)
  File "I:\Python\lib\site-packages\adafruit_blinka\microcontroller\pico_u2if\pico_u2if.py", line 325, in spi_write
    bytes([write_cmd, chunk]) + buffer[start : (start + chunk)], True
TypeError: can't concat list to bytes

Seems to be similar to #454 and changing displayio.display.py as intimated by @caternuson reply for that issue appears to fix it

 ```
pixels = list(
            numpy.dstack(((color >> 8) & 0xFF, color & 0xFF)).flatten().tolist()
        )
```
to
```
pixels = bytes(
           numpy.dstack(((color >> 8) & 0xFF, color & 0xFF)).flatten().tolist()
       )
```

caternuson, 2021-05-23:
The issue was actually in the RGB display library and should have been fixed with:
https://github.com/adafruit/Adafruit_CircuitPython_RGB_Display/pull/93
Are you using the latest release that includes the fix?
https://github.com/adafruit/Adafruit_CircuitPython_RGB_Display/releases/tag/3.10.7

cs0ckn, 2021-05-23:
I'm using `minitft_featherwing` from `adafruit_featherwing` library

`from adafruit_featherwing import minitft_featherwing`

so I don't have a dependency on `Adafruit_CircuitPython_RGB_Display` library as far as I can tell so I can't let you know which version of the RGB display library I use. If I don't make the change I described above I get the `bytes([write_cmd, chunk]) + buffer[start : (start + chunk)], True TypeError: can't concat list to bytes` exception each time I try to use the `display.show()` method from my `minitft_featherwing.MiniTFTFeatherWing` object instance. Any other pointers would be very grateful if this isn't a general problem.

caternuson, 2021-05-24:
@makermelissa Any ideas here? Could this be something in `Adafruit_Blinka_Displayio`? Or should something more like this approach be taken?
https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/tft-display

makermelissa, 2021-05-24:
Yeah @caternuson, mini_tftfeatherwing uses displayio, so it could definitely be Adafruit_Blinka_Displayio, especially since I referenced the RGB display library when designing it. In fact, I would think the same fix we did for the rgb library could be applied there as well.

caternuson, 2021-05-24:
Thanks. And @cs0ckn mentioned that at bottom of first post. Just now noticing that again :) OK, does seem like issue is that:
https://github.com/adafruit/Adafruit_Blinka_Displayio/blob/908fce087757d5d077cfceb7055047c02fbc793a/displayio/display.py#L266