adafruit/Adafruit_Blinka #480: Problem with spi and MicroPython

jerryneedell, 2021-05-30:
I ran into an issue trying to use Blinka under Micropython on a Raspberry Pi Pico. 
I was trying to see if I could use an "airlift" with the esp32spi library. 
The first problem had simple "workaround":
The esp32spi library uses time.monotonic() but time.monotonic() is not provided by the micropython time module... It looks like a simple fix is to just use time.time() use MP but it does mean a lot of CP code won't work without modification.. I did make that change and ran into some more significant issues that may indicate bigger problems, but I wanted to point out the time.monotonic issue to see if I was missing some thing simple to get around it.

The second issue may be more significant:
after replacing the time.monotonic() called with time.time() in adafruit_esp32spi.py
I ran into this
```
MPY: soft reboot
MicroPython v1.15 on 2021-04-18; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>> 
>>> 
>>> import esp32spi_simpletest
ESP32 SPI webclient test
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "esp32spi_simpletest.py", line 50, in <module>
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 335, in status
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 324, in _send_command_get_response
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 232, in _send_command
  File "adafruit_bus_device/spi_device.py", line 76, in __enter__
  File "/lib/busio.py", line 377, in configure
AttributeError: type object 'SPI' has no attribute 'MSB'
>>> 
```
Any suggestions on how to proceed?

ladyada, 2021-05-30:
we could probably fix the `MSB` thing. `time` is tougher. id actually recommend asking upstream micropython to add `monotonic` - we use it *a lot* in circuitpython code

jerryneedell, 2021-05-30:
Agreed -- it would be nice if MP supported time.monotonic()
This came up in response to a forum post https://forums.adafruit.com/viewtopic.php?f=53&t=179815  and I wanted to try it out myself before suggesting that Blinka could allow access to the AirLift.

makermelissa, 2021-05-30:
I think when testing, I used `time.ticks_ms() / 1000` in place of time.monotonic().

tannewt, 2021-06-01:
Please do not use `time.ticks_ms()` in CircuitPython code. It is a non-standard MicroPython API. `time.monotonic()` is used because it's available in CPython: https://docs.python.org/3/library/time.html#time.monotonic

I suspect we would make a PR to MicroPython to add both monotonic and monotonic_ns so more of the CP libraries work.

makermelissa, 2021-06-01:
@tannewt for submitting a PR to MP, where would the code be to add it? `shared-modules` and `shared-bindings`? Or other places too?

tannewt, 2021-06-02:
Looks like time is implemented per-port in MicroPython. Here is the STM one: https://github.com/micropython/micropython/blob/master/ports/stm32/modutime.c

makermelissa, 2021-06-02:
It appears the relevant ones may be these:
https://github.com/micropython/micropython/blob/master/ports/rp2/modutime.c
https://github.com/micropython/micropython/blob/master/ports/stm32/modutime.c
https://github.com/micropython/micropython/blob/master/ports/esp32/modutime.c
and possibly:
https://github.com/micropython/micropython/blob/master/ports/esp8266/modutime.c

However, they may want it to be consistent across ports.

elijahsgh, 2021-06-09:
> @dpgeorge Any tips on where to start? Thanks!

I'd be interested in trying to add this for the RP2040 specifically. Did you already have something started? No worries if not - I think I'll just try adding time_monotonic to the ports/rp2/moduetime.c files makermelissa listed.

makermelissa, 2021-06-09:
> I'd be interested in trying to add this for the RP2040 specifically. Did you already have something started? No worries if not - I think I'll just try adding time_monotonic to the ports/rp2/moduetime.c files makermelissa listed.

Awesome. I don't believe there's anything started yet.

putun0, 2021-10-06:
> I think when testing, I used `time.ticks_ms() / 1000` in place of time.monotonic().

Hi, I am trying to use esp32spi library too. I have the same problem. I changed 'time.monotonic()' to 'time.ticks_ms() / 1000' but still I have the same error.Any suggestions for get this library worked? Thank you.

putun0, 2021-10-06:
> > @dpgeorge Any tips on where to start? Thanks!
> 
> I'd be interested in trying to add this for the RP2040 specifically. Did you already have something started? No worries if not - I think I'll just try adding time_monotonic to the ports/rp2/moduetime.c files makermelissa listed.

Did you succeed ?

makermelissa, 2021-10-06:
> Hi @makermelissa , I am trying to use esp32spi library too. I have the same problem. I changed 'time.monotonic()' to 'time.ticks_ms() / 1000' but still I have the same error.Any suggestions for get this library worked? Thank you.

What is the error you are getting?

makermelissa, 2022-02-15:
As far as I know, it's always a float. If I had to guess, it would eventually wrap around after running for a prolonged time.

makermelissa, 2022-11-09:
I think this is heavily related to #7 (as a great idea for a potential fix).

makermelissa, 2025-08-22:
Using the Adafruit_CircuitPython_Ticks library should resolve this issue.

closed 2025-08-22 by makermelissa