adafruit/Adafruit_Blinka #482: Blink on Micropyton coul dnot read DP310 Sensor

jposada202020, 2021-06-03:
Doing some test, I found that the DPS310 is having with Blinkain Micriopython. The BME280 worked with any issue

### Hardware
```python
MicroPython v1.15 on 2021-04-18; Raspberry Pi Pico with RP2040
```

However I Used a different Sensor
I used an **Adafruit DPS310 Pressure Sensor** with the latest Blinka

### Results
While reading the sensor the following I got the following error message:
```python
Traceback (most recent call last):
  File "<stdin>", line 10, in <module>
  File "/lib/adafruit_dps310.py", line 228, in __init__
RuntimeError: Failed to find DPS310 - check your wiring!
```

Using the REPL to scan the sensor:
```python
>>> import board
>>> import busio
>>> i2c = busio.I2C(board.GP1, board.GP0)
>>> i2c.scan()
[119]
>>>
```

lesamouraipourpre, 2021-06-04:
I've found the cause but I'm not sure of the best fix.

Cause:
https://github.com/adafruit/Adafruit_Blinka/blob/78bfa5749837c033839d087a5aab7932960ea6f8/src/adafruit_blinka/microcontroller/rp2040/i2c.py#L58-L63
The read_buffer was read into with offset 0 but we are not copying from there.

My test fix which works:
```python
        read_buffer = buffer_in
        self.readfrom_into(address, read_buffer, stop=stop)
#        if in_end:
#            buffer_in[in_start:in_end] = read_buffer[in_start:in_end]
#        else:
#            buffer_in[in_start:] = read_buffer[in_start:]
        if in_end:
            buffer_in[in_start:in_end] = read_buffer[0:in_end-in_start]
        else:
            buffer_in[in_start:] = read_buffer[0:len(read_buffer)-in_start]
```

This is by no means tested, and I'm not sure what the edge cases are to test against.

makermelissa, 2021-06-04:
Sounds good. Want to submit a PR? It was probably just me not quite understanding the start/end numbers and it happened to work in my particular case, which was the bme280. This fix may also fix the adt7410 which I wasn't able to get working.

lesamouraipourpre, 2021-06-04:
I'll get one in today and we can work on testing it.

jposada202020, 2021-06-04:
Yes let me know I have both sensor I could help testing them

closed 2021-06-04 by makermelissa