adafruit/Adafruit_Blinka #485: Strange behaviour when using 3.5in piTFT with touch

Jayguitar, 2021-06-09:
When monitoring touch input with the built in stmpe610 resistive touch chip a strange sequence of pixels appear on the top left of the screen. Reproducible with the following code:

import board
import displayio
import busio
import digitalio
import terminalio
import adafruit_hx8357
from adafruit_stmpe610 import Adafruit_STMPE610_SPI
from adafruit_display_text import label

displayio.release_displays()
spi = board.SPI()
tft_cs = board.CE0
tft_dc = board.D25
rt_cs = digitalio.DigitalInOut(board.CE1) 
rt = Adafruit_STMPE610_SPI(spi, rt_cs)

display_bus = displayio.FourWire(spi, command=tft_dc, chip_select=tft_cs)
display = adafruit_hx8357.HX8357(display_bus, width=480, height=320)

splash = displayio.Group()
display.show(splash)

text = "HELLO WORLD"
font = terminalio.FONT
color = 0xFF00FF
 
text_area = label.Label(font, text=text, color=color)
 
text_area.x = 20
text_area.y = 40
 
splash.append(text_area)

while True:
    print(rt.touches)

Increasing the complexity of the code, for example adding buttons that should react to touch, results in even more bizarre behaviour on the screen.

Blinka 5.13.0
displayio 0.5.5
hx8357 1.2.4
stmpe 1.2.3

makermelissa, 2021-06-09:
I moved this to Blinka for now because the issue isn't with the Adafruit_Blinka_Displayio library itself.

I tried using Adafruit_Python_Extended_Bus to get around this issue because we don't need to make it guess spidev0.0 or spidev0.1, but I think there is a bit being added or lost somewhere. The chip ID is coming back as 0x408 instead of 0x811. In binary, this is `0100 0000 1000` instead of `1000 0001 0001`.

makermelissa, 2021-06-15:
Ok, I figured out a workaround to this for now. Raspberry Pi OS allows you to reassign the CS pins to other GPIO so it stops messing with the lines. Here's how to change it:
Open `/config/boot.txt`
Make sure `dtparam=spi=on` is commented out or it can be disabled through `raspi-config`
Add `dtoverlay=spi0-2cs,cs0_pin=5,cs1_pin=6` to the config.

The last line will re-assign those to GPIO 5 and 6 since they aren't used by the PiTFT, but you can change that to something else if desired. Using this method, I was able to use the original code unmodified without the strange behavior.

makermelissa, 2021-06-16:
I wrote a script to make this easier and have updated the SPI page on the Raspberry Pi guide with how to use it: https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/spi-sensors-devices