adafruit/Adafruit_Blinka #486: python 3.5 vs 3.8

plotti, 2021-06-11:
Cache entry deserialization failed, entry ignored
  Cache entry deserialization failed, entry ignored
  Using cached https://files.pythonhosted.org/packages/f3/95/8f6fe08d1f361c16f8ce384f85ea3b86739053c285b5db90dcc3c9d101d4/pyftdi-0.52.9-py3-none-any.whl
Collecting sysv-ipc (from Adafruit-Blinka->adafruit-circuitpython-bitbangio)
  Using cached https://files.pythonhosted.org/packages/0c/d7/5d2f861155e9749f981e6c58f2a482d3ab458bf8c35ae24d4b4d5899ebf9/sysv_ipc-1.1.0.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-u54gk5m5/sysv-ipc/setup.py", line 11, in <module>
        import prober
      File "/tmp/pip-build-u54gk5m5/sysv-ipc/prober.py", line 137
        d["SYSV_IPC_VERSION"] = f'"{version}"'
                                             ^
    SyntaxError: invalid syntax
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-u54gk5m5/sysv-ipc/

---> is it really necessary to have python 3.8 the raspbian comes with 3.5 . I dont want to update all of it, is there a workaround?

lesamouraipourpre, 2021-06-14:
Are you still running `Raspbian` or the renamed and up-to-date [`Raspberry Pi OS`](https://www.raspberrypi.org/software/)?
Blinka works fine for me on it:
```bash
python3 --version
Python 3.7.3
```
I don't believe there is anything that requires Python 3.8, Blinka lists 3.6 as a requirement:
https://github.com/adafruit/Adafruit_Blinka/blob/cf718220ada6c7665e8ab54c2d6ce1bc50d0b5da/setup.py#L44

makermelissa, 2021-06-14:
I think Python 3.5 came with the old Stretch release.

plotti, 2021-06-21:
I am running on an old strech release and even when i define that i want to install adafruit-blinka with an old version e.g. 3.0.0 i ge this below (aka it still wants to install ysv_ipc-1.1.0.tar.gz which is a blocker, since it only works with python > 3.5

pi@raspberrypi:~ $ pip3 install Adafruit-Blinka==3.0.0 --no-cache-dir
Collecting Adafruit-Blinka==3.0.0
  Downloading https://www.piwheels.org/simple/adafruit-blinka/Adafruit_Blinka-3.0.0-py3-none-any.whl (83kB)
    100% |████████████████████████████████| 92kB 1.6MB/s 
Collecting rpi-ws281x>=4.0.0 (from Adafruit-Blinka==3.0.0)
  Downloading https://www.piwheels.org/simple/rpi-ws281x/rpi_ws281x-4.3.0-cp35-cp35m-linux_armv7l.whl (106kB)
    100% |████████████████████████████████| 112kB 1.9MB/s 
Collecting Adafruit-PlatformDetect (from Adafruit-Blinka==3.0.0)
  Downloading https://www.piwheels.org/simple/adafruit-platformdetect/Adafruit_PlatformDetect-2.28.0-py3-none-any.whl
Collecting Adafruit-PureIO (from Adafruit-Blinka==3.0.0)
  Downloading https://www.piwheels.org/simple/adafruit-pureio/Adafruit_PureIO-1.1.9-py3-none-any.whl
Collecting RPi.GPIO (from Adafruit-Blinka==3.0.0)
  Downloading https://www.piwheels.org/simple/rpi-gpio/RPi.GPIO-0.7.0-cp35-cp35m-linux_armv7l.whl (67kB)
    100% |████████████████████████████████| 71kB 1.3MB/s 
Collecting spidev; sys_platform == "linux" (from Adafruit-Blinka==3.0.0)
  Downloading https://www.piwheels.org/simple/spidev/spidev-3.5-cp35-cp35m-linux_armv7l.whl
Collecting sysv-ipc; platform_system != "Windows" (from Adafruit-Blinka==3.0.0)
  Downloading https://files.pythonhosted.org/packages/0c/d7/5d2f861155e9749f981e6c58f2a482d3ab458bf8c35ae24d4b4d5899ebf9/sysv_ipc-1.1.0.tar.gz (99kB)
    100% |████████████████████████████████| 102kB 4.1MB/s 
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-wtwnocb5/sysv-ipc/setup.py", line 11, in <module>
        import prober
      File "/tmp/pip-build-wtwnocb5/sysv-ipc/prober.py", line 137
        d["SYSV_IPC_VERSION"] = f'"{version}"'
                                             ^
    SyntaxError: invalid syntax
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-wtwnocb5/sysv-ipc/

makermelissa, 2021-06-21:
I think you would need both PlatformDetect 2.28.0 and Blinka 5.13.1, which are the releases immediately before we changed to needing Python 3.6.

plotti, 2021-06-21:
pip3 install Adafruit-Blinka==3.0.0 --no-cache-dir --> but why doesn't this automatically have the right dependencies, it shouldn't need sysv_ipc-1.1.0.tar.gz as it does there ... i am puzzled....

plotti, 2021-06-21:
[image]

i get it the sysipv version is not defined in the older packages see screenshot

makermelissa, 2021-06-23:
That makes sense, which means it just installs the latest version. You might be able to get the latest version to install locally by downloading the Blinka and PlatformDetect repos to your device, changing the version in the setup.py files and running `pip3 install -e .` inside the folder of each.

makermelissa, 2021-09-13:
Closing because we don't have any plans to go back to Python 3.5.

closed 2021-09-13 by makermelissa