adafruit/Adafruit_Blinka #487: GENERIC_X86 - Microcontroller not supported

lesamouraipourpre, 2021-06-14:
Ref: https://github.com/adafruit/Adafruit_Blinka_Displayio/actions/runs/936869211 (if it's not been deleted)
PR: https://github.com/adafruit/Adafruit_Blinka_Displayio/pull/58
```
  File "/home/runner/work/Adafruit_Blinka_Displayio/Adafruit_Blinka_Displayio/displayio/i2cdisplay.py", line 30, in <module>
    import microcontroller
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/microcontroller/__init__.py", line 117, in <module>
    raise NotImplementedError("Microcontroller not supported:", chip_id)
NotImplementedError: ('Microcontroller not supported:', 'GENERIC_X86')
```

When running Github Actions, in the example above to build documentation, that uses `microcontroller` it will fail because there is not a fall-back case of `GENERIC_X86`.

Should this be added?
Or does this open up a whole can-of-worms that we don't want to get into?

In the example above, it can be fixed by removing the paramter typing that references `microcontroller.Pin`.

makermelissa, 2021-06-14:
I'll take a look. I don't think this is being detected correctly...

makermelissa, 2021-06-14:
Ok, I think https://github.com/adafruit/Adafruit_Blinka/blob/main/src/microcontroller/__init__.py needs something like https://github.com/adafruit/Adafruit_Blinka/blob/main/src/board.py#L242-L243 added to it. I'll make a PR.

PR #488 (2021-06-14): Fix docs not passing if microcontroller is imported
Fixes #487.
closed 2021-06-14 by makermelissa