adafruit/Adafruit_Blinka #489: NeoPixel only working on GPIO18

spike322, 2021-06-22:
Hi, I'm developing an application using a Raspberry Pi CM4 in which I need to use a 12 led RGB WS2812B ring, and more specifically I have to create an animation where all of the 12 leds fade from minimum brightness to max and vice versa. I noticed that if I control the leds using GPIO18 it works fine, but if I try using a different pin, like GPIO10 or GPIO21, it has strange behaviours. For instance, all the leds become white and even if I send a different colour they keep staying white, or if I dimmer them from min to max, while the transition, different leds blink with different colours, completely at a random.

Here's reported a snippet from the code I'm using.

```python
import time
import board
import neopixel

pixel_pin = board.D21

num_pixels = 12

ORDER = neopixel.GRB

pixels = neopixel.NeoPixel(
    pixel_pin, num_pixels, brightness=0.2, auto_write=False, pixel_order=ORDER
)

try:
    while True:
        for i in range(1, 255):
            pixels.fill((i, 0, 0))
            pixels.show()
            time.sleep(0.005)
        
        for i in range(254, 2, -1):
            pixels.fill((i, 0, 0))
            pixels.show()
            time.sleep(0.005)
        
except KeyboardInterrupt:
    print('Keyboard interrupt')
    pixels.fill(0)
    pixels.show()
    pixels.deinit()
```

As said before, during the transition I'm actually facing leds changing colours and blinking completely at a random. This happens only if I use a GPIO that is not GPIO18. If I use GPIO18 for controlling the led ring everything works fine. I even tested the same code with two different OS (Raspberry OS and Manjaro ARM) but I have more or less the same behaviour.

spike322, 2021-06-22:
While trying different things I noticed that if I have the `pigpiod` daemon running I got strange behaviours on my rgb leds, if I kill it they seem to work normally. But in my case I need `pigpiod` and `neopixel` working together...

2bndy5, 2021-07-02:
Last I checked, the raspberry pi only has 2 channels of hardware controlled PWM; channel0 is connected to GPIO18 & GPIO19, and channel1 is connected to GPIO12 & GPIO13 - these channels might also be tied to the analog 3.5mm audio output connector.

I think this issue is more related to the hardware limitation rather than a software limitation. The pigpio lib is unique in that it boasts hardware timed PWM, but I'm not familiar enough with it's source to definitively say why it would cause "strange behavior". My guess is that pigpiod is trying to use a PWM channel while your app is also trying to use the same channel.

spike322, 2021-07-07:
> Last I checked, the raspberry pi only has 2 channels of hardware controlled PWM; channel0 is connected to GPIO18 & GPIO19, and channel1 is connected to GPIO12 & GPIO13 - these channels might also be [...]

Thank you for your reply! So you think there's no way to make them working together without conflict?

2bndy5, 2021-07-07:
I'm not an expert, its just my guess. Your issue seems like it needs more research. I think that [this is the code called by this lib's `neopixel.show()` on the BCM283x (RPi)](https://github.com/adafruit/Adafruit_Blinka/blob/a79eb436569f2c4d1390571eba5779fc5aedf6db/src/adafruit_blinka/microcontroller/bcm283x/neopixel.py#L22). But Again I'm not familiar with the source code for pigpio...

makermelissa, 2021-07-30:
> Last I checked, the raspberry pi only has 2 channels of hardware controlled PWM; channel0 is connected to GPIO18 & GPIO19, and channel1 is connected to GPIO12 & GPIO13 - these channels might also be [...]

I think when I was looking into a similar issue a while back, this is the conclusion I came to as well. It's not shareable and they're essentially quantum entangled and modifying one modifies the other.

makermelissa, 2021-09-10:
Closing because this is really a hardware limitation and nothing to do here.

closed 2021-09-10 by makermelissa