adafruit/Adafruit_Blinka #497: type object 'gpiod.Chip' has no attribute 'OPEN_BY_NAME' on Orange Pi Zero

mytja, 2021-07-26:
Hello!
I recently used your library, and it's amazing. But, sadly I cannot import board library. I am using Orange Pi Zero on latest Armbian with Python 3.7.3.
Here are logs when I import board library
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/dist-packages/board.py", line 96, in <module>
    from adafruit_blinka.board.orangepi.orangepizero import *
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/board/orangepi/orangepizero.py", line 5, in <module>
    from adafruit_blinka.microcontroller.allwinner.h3 import pin
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/allwinner/h3/pin.py", line 4, in <module>
    PA0 = Pin(0)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_blinka/microcontroller/generic_linux/libgpiod_pin.py", line 34, in __init__
    self._chip = gpiod.Chip("gpiochip0", gpiod.Chip.OPEN_BY_NAME)
AttributeError: type object 'gpiod.Chip' has no attribute 'OPEN_BY_NAME'
```
I have built gpiod library using provided script: https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/libgpiod.sh
Any clue of what is going wrong?
Thank you so much

makermelissa, 2021-07-26:
You may have better luck trying to install with apt-get. Here's how:
https://learn.adafruit.com/adding-a-single-board-computer-to-blinka/software-setup#installing-the-libgpiod-as-a-package-3058813-3

mytja, 2021-07-26:
Hello and thanks for your fast respose!
I will test it tomorrow, and I hope it will work.
Again, thanks and have a nice day.

mytja, 2021-07-28:
It works very well.
Thanks for your time, I appreciate it.

closed 2021-07-28 by mytja