adafruit/Adafruit_Blinka #499: misleading docstr in DigitalInOut.pull

2bndy5, 2021-07-30:
https://github.com/adafruit/Adafruit_Blinka/blob/4f4e6af0a2c2912d7662f0160849806507b125bd/src/digitalio.py#L190-L191

I think this docstr should say
```py
"""Get out set the Digital Pin's input impedance."""
```

ps - I think this is simply a copy-n-paste mistake.

makermelissa, 2021-07-30:
Possibly. It's been there longer than I've been working on this project.

tannewt, 2021-08-02:
Property documentation shouldn't say `get or set` that is implied. The CircuitPython doc string for this is `The pin pull direction.` https://github.com/adafruit/circuitpython/blob/main/shared-bindings/digitalio/DigitalInOut.c#L306

makermelissa, 2021-08-02:
It should probably be updated to `"""Get or Set the Digital Pin Pull Direction"""`

2bndy5, 2021-08-02:
@tannewt is right. The "get or set" not being needed is a  stipulation actually noted in the CirPy contribution guide. I was just guessing with my first suggestion.

makermelissa, 2021-08-02:
Oh I see, I've probably made that mistake in other libraries myself.

PR #505 (2021-09-10): Update digitalio docstrings to be more in line with CircuitPython
Fixes #499.
closed 2021-09-10 by makermelissa