adafruit/Adafruit_Blinka #500: Jetson Nano GEN1 v GEN2 I2C pin mappings

lackdaz, 2021-08-03:
https://github.com/adafruit/Adafruit_Blinka/blob/4f4e6af0a2c2912d7662f0160849806507b125bd/src/adafruit_blinka/microcontroller/tegra/t210/pin.py#L82

Does anyone know what's the difference between "GEN1_I2C_SDA" vs "GEN2_I2C_SDA" pin mappings?

@makermelissa 

I've had i2c code that initially worked with the GEN1 mappings but got an IOError.

Changing to "GEN2_I2C_SDA"/"GEN2_I2C_SCL" or using `SDA_1`/`SCL_1` solved the issue but I'm wondering if this is a board revision between the a2 and b01, or specific to the l4t tegra version.

lackdaz, 2021-08-03:
Ok I think I got it, I inferred from your [initial setup guide](https://learn.adafruit.com/circuitpython-libraries-on-linux-and-the-nvidia-jetson-nano/initial-setup) that these refer to separate channels?

`SDA_1` - refers to board pins 3/5 (Bus 1)
`SDA` - refers to board pins 27/28 (Bus 0)

Do close the issue if this is confirmed!

makermelissa, 2021-08-19:
Yeah, looking at the boards, this sounds correct.

closed 2021-08-19 by makermelissa