adafruit/Adafruit_Blinka #507: [Feature request] Implement configurable pin functions (LED_URX, LED_UTX, LED_I2C) on MCP2221

fijam, 2021-09-13:
MCP2221 datasheet mentions alternate pin functions in the pin mux module such as LED_URX, LED_UTX and LED_I2C which can be used for visual indication of UART/I2C traffic.

[image]

These are available on GP0, GP1 and GP3. It would be nice to be able to take advantage of those,

[image]

fijam, 2021-09-24:
I dug around a bit and figured out that alternative functions can be enabled manually. For example, for LED_I2C on GP3:

```    
from adafruit_blinka.microcontroller.mcp2221 import mcp2221
mcp2221.mcp2221.gp_set_mode(3, 0b001)
```

caternuson, 2021-09-24:
I'm not sure how we could expose this in the context of Blinka. @makermelissa any ideas?

makermelissa, 2021-09-24:
Yeah, sort of for i2c at least. The way I added the Pico is I added all of the pin combos in the microcontroller file. For the mcp2221, with the i2c implementation, you could figure out if it's an alternate i2c port and if so, just set the mode in the microcontroller specific implementation of i2c.

closed 2026-10-20 by makermelissa